geeekpi/upsplus #4: battery is never reaching FULL status

deHarro, 2021-04-18:
I attached the UPS Plus V5 to a Raspberry 3B+.
My batteries have a inbuild security chip (preventing the battery from beeing depleted too deep).
The power supply (PS) is a genuine Raspberry 4 PS delivering 3 A@ 5.1 V.

After many hours running from the PS, the fourth LED of the UPS is still blinking and the batteries are charged @~40 mA continuously.
Reading the registers for "Full Voltage", Protection Voltage" and "Empty Voltage" several times in series, I realized, that the "Full Voltage" register is always changing to higher values (starting with around 4000, going up to 4289).
Those higher values can never be reached by the battery, the battery voltage is around 4208, which is a sane value for a Lipo battery.

I expected, that after max 3..4 hours the batteries are judged full and from then on the PS should only supply the RPi.

I tried to lower the register for "Full Voltage" to 4200, knowing that this value will be reached by the battery, but the controller of the UPS changes this value permanently.

Is this behavior on purpose or is it a bug?
Thanks!

Harry

nickfox-taterli, 2021-04-20:
[Full Voltage] means the maximum voltage of the battery ever sampled, which will be recharged at intervals since it will be slowly discharged afterwards.

deHarro, 2021-04-20:
I realized, that the controller is permanently switching between "charging" and "discharging" with very low currents.
So it is to be considered normal, that the fourth blue LED is never becoming statically ON, as long as a power supply is connected?

Thanks for clarifying that. You should mention that in the documentation.

closed 2021-04-20 by deHarro