geeekpi/upsplus #7: AC power coming back during shutdown countdown

frtz13, 2021-04-28:
using firmware v. 4
when AC power comes back on during the shutdown countdown, I realized that the 5V supply for the raspberry pi is not switched off at the end of the countdown. so, after shutdown of the operating system, the raspberry pi will never restart automatically.
IMHO the following would be better: once a shutdown countdown starts, and we have the Back-To-AC Auto Power UP flag set, then:
- the shutdown sequence should proceed to its end,
- switch off the 5V-power for a certain time (a couple of seconds maybe), regardless of AC Power.
- switch the 5V back on when AC power is restored.

otherwise, I am very happy with the product.
one more thing: would be nice to have some sort of alert for new firmware releases and release notes for these.

nickfox-taterli, 2021-05-02:
We are making changes to the firmware for this feature, but we still need some internal testing before release, which is expected to be done within a week.

nickfox-taterli, 2021-05-06:
The test is being conducted by @yoyojacky .

yoyojacky, 2021-05-06:
> using firmware v. 4
> when AC power comes back on during the shutdown countdown, I realized that the 5V supply for the raspberry pi is not switched off at the end of the countdown. so, after shutdown [...]

Thank you for your approval. We are updating the firmware. Now the firmware version is v5. You can pull the latest python script file by typing git pull in the warehouse directory, and then please follow the instructions on the wiki to perform the OTA upgrade. After the upgrade, you can download The power will be cut off after the countdown is over. please pay attention to the countdown time default is 240 seconds, you can modify it by Editing Full-featured-demo-code.py. 
or you can just do it as following step:

Step1.  Plug the power supply. make sure it charging your batteries.

Step2. Update firmware via OTA  by typing following command in terminal:
```bash
i2cset -y 1 0x17  50 127  b
sudo sync && sudo init 0
```
Step3. Remove the power and batteries and reinsert batteries back to UPS and plug the power supply again.
Step4. Enter into upsplus folder where you cloned the repository and typing following command : 
```bash
python3 OTA_firmware_upgrade.py
```
it will shutdown your Pi automatically.
Step5. Remove the batteries and power and reinsert it back to your UPS, plug the power supply again.
Step6.  Check if the register is 0x17 in terminal :  
```bash
i2cdetect -y 1  
```
you would see:
<pre>
pi@upstest:~/upsplus $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- 17 -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: 40 -- -- -- -- 45 -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
</pre>
it means your UPS is ok for working..
Step 7.  Setting the function of `back-to-ac`: 
```bash
i2cset -y 1 0x17  25 1 
```
Step 8.  Setting `shutdown Countdown` function:
```bash
i2cset -y  1 0x17  24  20 
``` 
PS: 20 means Counting down in 20 seconds. 
Step 9.  Watch and wait
That's all, thank you~

closed 2021-05-06 by yoyojacky

mmaga, 2021-12-11:
> 

I've been experiencing "no countdown" after setting the UPS to shutdown after 20 seconds (tried 10 and 15 also), but the UPS was not counting down.

So, I upgraded to the last firmware following @yoyojacky instructions, and I got stuck in OTA Firmware Upgrade Mode (0x18).

May you help, please?

mmaga, 2021-12-11:
> I've been experiencing "no countdown" after setting the UPS to shutdown after 20 seconds (tried 10 and 15 also), but the UPS was not counting down.
> 
> So, I upgraded to the last firmware following @yoyojacky instructions, and I got stuck in OTA Firmware Upgrade Mode (0x18).
> 
> May you help, please?

0x17 Operation mode is back! 
I removed the option i2c_arm_baudrate=400000 from the /boot/config.txt.
It may be useful to someone else.

yoyojacky, 2022-03-02:
> > I've been experiencing "no countdown" after setting the UPS to shutdown after 20 seconds (tried 10 and 15 also), but the UPS was not counting down.
> > So, I upgraded to the last firmware following @yoyojacky instructions, and I got stuck in OTA Firmware Upgrade Mode (0x18).
> > May you help, please?
> 
> 0x17 Operation mode is back, and shutdown countdown also! I removed the option i2c_arm_baudrate=400000 from the /boot/config.txt. It may be useful to someone else.. but the way, which is the suggested baudrate for UPS Plus?

default is good, no need to change the value of the i2c_baudrate.