geeekpi/upsplus #11: Battery voltage measurement is sporad

Dulus0, 2021-05-05:


Dulus0, 2021-05-05:
Battery voltage measurement is working intermittently

These two measurement were done directly after each other.
This means, that the RPi will shutdown as soon as i unplug the power adapter.

RPi B+ with latest Raspian fully patched.

pi@raspberrypi:~$ python3 ~/bin/upsPlus.py
------------------------------------------------------------
------Current information of the detected Raspberry Pi------
------------------------------------------------------------
Raspberry Pi Supply Voltage: 4.984 V
Raspberry Pi Current Current Consumption: 622.035 mA
Raspberry Pi Current Power Consumption: 3125.315 mW
------------------------------------------------------------
-------------------Batteries information-------------------
------------------------------------------------------------
Voltage of Batteries: 4.240 V
Battery Current (Charging) Rate: 131.951 mA
Current Battery Power Supplement: 0.000 mW
Successfully set the protection voltage to: 3700 mV
------------------------------------------------------------
Currently charging via Micro USB Port.
pi@raspberrypi:~$ 
pi@raspberrypi:~$ 
pi@raspberrypi:~$ 
pi@raspberrypi:~$ python3 ~/bin/upsPlus.py
------------------------------------------------------------
------Current information of the detected Raspberry Pi------
------------------------------------------------------------
Raspberry Pi Supply Voltage: 4.984 V
Raspberry Pi Current Current Consumption: 631.623 mA
Raspberry Pi Current Power Consumption: 3118.587 mW
------------------------------------------------------------
-------------------Batteries information-------------------
------------------------------------------------------------
Voltage of Batteries: 0.000 V
Battery Current (Charging) Rate: 128.049 mA
Current Battery Power Supplement: 0.000 mW
Successfully set the protection voltage to: 3700 mV
------------------------------------------------------------
Currently charging via Micro USB Port.
pi@raspberrypi:~$

nickfox-taterli, 2021-05-06:
You can upgrade the firmware and try again, or contact the aftermarket and provide the serial number for inspection.

Please also provide a photo of the front of the product, connected to the Raspberry Pi and a separate picture to check if it is a faulty product.

closed 2021-05-06 by nickfox-taterli