geeekpi/upsplus #15: ups switches back on after shutdown countdown, even when AC power is not restored

frtz13, 2021-05-06:
follow-up on issue #7 

using firmware v. 5
after a shutdown countdown, with Back-To-AC Auto Power up enabled, the UPS switches back on, even when no AC power supply is connected.

pi@raspberrypi:~/upsplus $ python3 Full-featured-demo-code.py
Raspberry Pi power supply voltage: 4.936 V
Current current consumption of Raspberry Pi: 1184.861 mA
Current power consumption of Raspberry Pi: 5665.265 mW
Batteries Voltage: 4.064 V
Battery current (discharge), rate: 1590.000 mA
Current battery power consumption: 6473.171 mW
Current processor voltage: 3282 mV
Current Raspberry Pi report voltage: 4962 mV
Current battery port report voltage: 4042 mV
Current charging interface report voltage (Type C): 73 mV
Current charging interface report voltage (Micro USB): 0 mV
Currently not charging.
Current battery temperature (estimated): 46 degC
Full battery voltage: 4285 mV
Battery empty voltage: 3700 mV
Battery protection voltage: 3700 mV
Battery remaining capacity: 55 %
Sampling period: 2 Min
Current power state: normal
No shutdown countdown!
Automatically turn on when there is external power supply!
No restart countdown!
Accumulated running time: 7634 sec
Accumulated charged time: 7639 sec
This running time: 5964 sec
Version number: 5
Serial Number is:(...)

nickfox-taterli, 2021-05-06:
You may need to update the firmware via [Method 1](https://wiki.52pi.com/index.php/UPS_Plus_SKU:_EP-0136#Method_1), which will make him forget the configuration that already exists, and look forward to the next feedback.

frtz13, 2021-05-06:
Firmware had been updated via Method 1. I took out the batteries another time for reset. still the same fault.

nickfox-taterli, 2021-05-07:
Please let us know the process of testing and we will try to reproduce this issue.

frtz13, 2021-05-07:
steps to reproduce the issue:
- take batteries out and put back in (make sure to have a known state). raspberry pi starts.
- Supply AC power.
- run demo code (1)
- switch off AC power
- configure back-to-AC (already set, but do it again): i2cset -y 1 0x17  25 1
- run demo code (2)
- start shutdown countdown: i2cset -y  1 0x17  24  30
- sudo poweroff
after 30 s, UPS switches off the 5V-output for a couple of seconds and then switches back on. Raspberry Pi reboots.

demo code result (1):
pi@raspberrypi:~/upsplus $ python3 Full-featured-demo-code.py
Raspberry Pi power supply voltage: 4.928 V
Current current consumption of Raspberry Pi: 1082.759 mA
Current power consumption of Raspberry Pi: 5739.277 mW
Batteries Voltage: 4.212 V
Battery current (charging), rate: 201.951 mA
Current battery power supplement: 936.585 mW
Current processor voltage: 3206 mV
Current Raspberry Pi report voltage: 4855 mV
Current battery port report voltage: 4100 mV
Current charging interface report voltage (Type C): 4828 mV
Current charging interface report voltage (Micro USB): 0 mV
Currently charging through Type C.
Current battery temperature (estimated): 54 degC
Full battery voltage: 4285 mV
Battery empty voltage: 3700 mV
Battery protection voltage: 3700 mV
Battery remaining capacity: 74 %
Sampling period: 2 Min
Current power state: normal
No shutdown countdown!
Automatically turn on when there is external power supply!
No restart countdown!
Accumulated running time: 171 sec
Accumulated charged time: 171 sec
This running time: 171 sec
Version number: 5
Serial Number is:(…)

demo-code result  (2)
pi@raspberrypi:~/upsplus $ python3 Full-featured-demo-code.py
Raspberry Pi power supply voltage: 4.932 V
Current current consumption of Raspberry Pi: 1089.655 mA
Current power consumption of Raspberry Pi: 5359.125 mW
Batteries Voltage: 4.140 V
Battery current (discharge), rate: 1585.854 mA
Current battery power consumption: 6848.780 mW
Current processor voltage: 3301 mV
Current Raspberry Pi report voltage: 4988 mV
Current battery port report voltage: 4138 mV
Current charging interface report voltage (Type C): 254 mV
Current charging interface report voltage (Micro USB): 3 mV
Currently not charging.
Current battery temperature (estimated): 42 degC
Full battery voltage: 4285 mV
Battery empty voltage: 3700 mV
Battery protection voltage: 3700 mV
Battery remaining capacity: 56 %
Sampling period: 2 Min
Current power state: normal
No shutdown countdown!
Automatically turn on when there is external power supply!
No restart countdown!
Accumulated running time: 897 sec
Accumulated charged time: 897 sec
This running time: 897 sec
Version number: 5
Serial Number is:(…)

nickfox-taterli, 2021-05-10:
We have made some updates to the software that seem to have solved the problem, please test it and wait for your feedback.

frtz13, 2021-05-10:
using firmware version 6.
the typical scenario now works again:
- switch AC power off
- start shutdown countdown and shutdown Raspberry Pi
- at the end of the countdown, UPS switches output power off and stays off until AC power is restored.

However, the following scenario (issue #7) still does _not_ work:
- switch AC power off
- start shutdown countdown and shutdown Raspberry Pi
- restore AC power
- at the end of the countdown, the UPS does _not_ switch off the output voltage for a couple of seconds. so the Raspberry Pi will never boot up again without user intervention (cut AC power, press button to switch off the UPS, then restore AC power).

nickfox-taterli, 2021-05-12:
The problem has been identified and a firmware update has been made, please update to version 7 and test.

frtz13, 2021-05-12:
using firmware version 7.
both of the above scenarios now work fine.
thanks.

closed 2021-05-12 by frtz13