geeekpi/upsplus #16: [Feature Request] Maximum battery charge limit

leandroalbero, 2021-05-09:
Hello, lots of phone or laptop manufacturers are including optimised battery charging into their devices. What this does is halt charge at around 80% instead of holding the battery cells at 100% charge during long periods of time. This greatly reduces battery ageing. 
Is it possible to do this via software on upsplus?

nickfox-taterli, 2021-05-09:
At the moment is to try to keep it around 95%, that is why always see in the charge, we are considering a more optimal battery protection method, if you have reference to the open source project, you can also recommend us.

leandroalbero, 2021-05-09:
[image]
Maybe it is possible to set the battery type (VSET) via the raspberry pi and thus lower the charge threshold. Don't know much about how this product is built, I was just asking if theres any chance of limiting the maximum charge via software.
Thanks!

nickfox-taterli, 2021-05-10:
Due to the characteristics of the battery, it will be used for the first time as heavy as possible to the highest voltage (charging current of less than 10mA considered full), and then dynamically set the VSET, we plan to design a function that can use software control VSET, which is supported in the hardware design (PCB Version > PCB01b), which generally affects the 80% to 100% charging stage.

nickfox-taterli, 2021-05-12:
We have developed a feature(beta), please update the firmware to version 7, to add a new register FixedVbat (Reg Addr:0x2A), when set to 1, the Full Voltage and Empty Voltage registers are(must) manually edited,when FixedVbat is 1,make the battery Full Voltage always lower than Full Voltage, but Full Voltage only can be set to 4200mV or upper, which is the charge management chip lower-limit.

tokp, 2021-05-15:
Could you put a example to write and read these values?

nickfox-taterli, 2021-05-18:
After you execute "bus.write_byte_data(DEVICE_ADDR, 42, 1)", you can adjust the [ Full Voltage and Empty Voltage registers ] registers manually, if you execute "bus.write_byte_data(DEVICE_ADDR, 42, 0)", the firmware will adjust them automatically, the automatic adjustment means that the battery maximum voltage is full and the minimum voltage is empty, regardless of the automatic or manual setting, the battery enters trickle charge mode when it reaches 90%.

Serhii020472, 2021-05-19:
Updated to version 7 and.... not changed [Full Voltage] [Empty Voltage]

pi@raspberrypi:~/bin $ i2cget -y 1 0x17 0x28 w
0x0007
pi@raspberrypi:~/bin $ i2cget -y 1 0x17 0x2A b
0x01
pi@raspberrypi:~/bin $ i2cget -y 1 0x17 0x0D w
0x1082
pi@raspberrypi:~/bin $ i2cget -y 1 0x17 0x0F w
0x102e
pi@raspberrypi:~/bin $ i2cget -y 1 0x17 0x11 w
0x0e10
pi@raspberrypi:~/bin $ i2cset -y 1 0x17 0x0D 0x1078 w
pi@raspberrypi:~/bin $ i2cset -y 1 0x17 0x0F 0x0D48 w
pi@raspberrypi:~/bin $ i2cset -y 1 0x17 0x11 0x0F55 w
pi@raspberrypi:~/bin $ i2cget -y 1 0x17 0x0D w
0x1082
pi@raspberrypi:~/bin $ i2cget -y 1 0x17 0x0F w
0x102e
pi@raspberrypi:~/bin $ i2cget -y 1 0x17 0x11 w
0x0f55

deHarro, 2021-05-21:
@Serhii020472
Can you please edit your posting to show all lines as they are (without strike through).
To achive this you should select all your text and then click on teh icon with "<>" in the caption bar of the edit box.

Like this: 
[image]

Thanks!
Harry

deHarro, 2021-05-21:
Hi @nickfox-taterli 
I updated my UPS to FW 7.
After reboot I wrote a short Python script to set the voltages as you described above.

```
import smbus2

DEVICE_BUS = 1
DEVICE_ADDR = 0x17
SAMPLE_TIME = 1

# new values for Full Voltage and Empty Voltage parameters
FULLVOLTAGE = 4200
EMPTYVOLTAGE = 3700

bus = smbus2.SMBus(DEVICE_BUS)

# enable writing to Full Voltage and Empty Voltage registers
bus.write_byte_data(DEVICE_ADDR, 42,1)

# write to the now enabled registers for Full Voltage and Empty Voltage
print("Set Full Voltage register to: %d" % FULLVOLTAGE)
bus.write_word_data(DEVICE_ADDR, 13, FULLVOLTAGE)
print("Set Empty Voltage register to: %d" % EMPTYVOLTAGE)
bus.write_word_data(DEVICE_ADDR, 15, EMPTYVOLTAGE)

# check result
print("FixedVbat register is: %d" %  bus.read_byte_data(DEVICE_ADDR, 42))

print("Full Voltage register is: %d" % bus.read_word_data(DEVICE_ADDR, 13))
print("Empty Voltage register is: %d" % bus.read_word_data(DEVICE_ADDR, 15))
```
The corresponding output is:
```
~/bin $ python set_Vfull_Vempty.py
Set Full Voltage register to: 4200
Set Empty Voltage register to: 3700
FixedVbat register is: 1
Full Voltage register is: 4293
Empty Voltage register is: 3601

```

The voltages are not altered, they remain as set by the firmware.

Is the procedure I implemented in my script correct?
Thanks!

Harry

deHarro, 2021-05-26:
@nickfox-taterli 
Any idea on that?

Harry

leandroalbero, 2021-05-26:
> Hi @nickfox-taterli
> I updated my UPS to FW 7.
> After reboot I wrote a short Python script to set the voltages as you described above.
> 
> ```
> # set_Vfull_Vempty.py
> 
> import smbus2
> 
> DEVICE_BUS = 1
> DEVICE_ADDR = 0x17
> SAMPLE_TIME = 1
> 
> # new values for Full Voltage and Empty Voltage parameters
> FULLVOLTAGE = 4200
> EMPTYVOLTAGE = 3700
> 
> bus = smbus2.SMBus(DEVICE_BUS)
> 
> # enable writing to Full Voltage and Empty Voltage registers
> bus.write_byte_data(DEVICE_ADDR, 42,1)
> 
> # write to the now enabled registers for Full Voltage and Empty Voltage
> print("Set Full Voltage register to: %d" % FULLVOLTAGE)
> bus.write_word_data(DEVICE_ADDR, 13, FULLVOLTAGE)
> print("Set Empty Voltage register to: %d" % EMPTYVOLTAGE)
> bus.write_word_data(DEVICE_ADDR, 15, EMPTYVOLTAGE)
> 
> # check result
> print("FixedVbat register is: %d" %  bus.read_byte_data(DEVICE_ADDR, 42))
> 
> print("Full Voltage register is: %d" % bus.read_word_data(DEVICE_ADDR, 13))
> print("Empty Voltage register is: %d" % bus.read_word_data(DEVICE_ADDR, 15))
> ```
> 
> The corresponding output is:
> 
> ```
> ~/bin $ python set_Vfull_Vempty.py
> Set Full Voltage register to: 4200
> Set Empty Voltage register to: 3700
> FixedVbat register is: 1
> Full Voltage register is: 4293
> Empty Voltage register is: 3601
> ```
> 
> The voltages are not altered, they remain as set by the firmware.
> 
> Is the procedure I implemented in my script correct?
> Thanks!
> 
> Harry
Hello, same result here:
```
pi@raspberrypi:~/upsplus$ python3 prueba.py 
Set Full Voltage register to: 4200
Set Empty Voltage register to: 3700
FixedVbat register is: 1
Full Voltage register is: 4284
Empty Voltage register is: 3700
```

nickfox-taterli, 2021-05-28:
assign to @yoyojacky  for testing.

leandroalbero, 2021-06-01:
[image]
I am using NCR18650B and according to the data-sheet its discharge-end voltage is 2.5V, I set the UPSPLUS (soft v8) empty voltage register to 3.2v to get more mAh out of the battery, but noticed that once it gets to 3.35v the battery voltage is incorrect. I measured voltage with a multimeter and it dropped as low as 2.5v. Is this a limitation of the voltmeter on the board?

ArjenR49, 2021-06-01:
I have the same sort of batteries and in v.7 or earlier there was no
problem going so low and have the ups Shut off when the batteries got
discharged. I would say it is not a hardware limitation. V.8 brought
changes but not for the better ...
Type journalctl -f in a terminal and have a good look there.

Arjen
(On The Road)

leandroalbero, 2021-06-01:
> I have the same sort of batteries and in v.7 or earlier there was no problem going so low and have the ups Shut off when the batteries got discharged. I would say it is not a hardware limitation. [...]

Have tried using journalctl -f while using i2cset, but I can't find any errors related to that command. 
Setting 0x10 register to 0x0c and confirming value not changed:
```
pi@raspberrypi:~$ i2cset -y 1 0x17 0x10 0x0c
pi@raspberrypi:~$ i2cget -y 1 0x17 0x10 b
0x00
```
journalctl output after above i2cset:
```
Jun 01 15:31:03 raspberrypi CRON[752]: (CRON) info (No MTA installed, discarding output)            
Jun 01 15:31:03 raspberrypi CRON[752]: pam_unix(cron:session): session closed for user pi           

```

nickfox-taterli, 2021-06-01:
- I just tested with the script provided in our repo, set the protection voltage 3700 mV, the battery cut off smoothly.
- Then change the protection to 3500 mV (readback to check) and run `stress -c 4` to pressurize the system.
- reach the protection voltage, smoothly cut off the power, using a multimeter to measure the voltage, indicating that the battery has 3.51V.

In addition, the protection voltage is determined by the protection voltage register, and the output will be turned off when the protection voltage is reached, the minimum voltage is used for battery estimation, not the shutdown function.

Please set the registers in order, for example, 0x0F and 0x10 are a group.

If your goal is to use more power, set a lower protection voltage as your request.

leandroalbero, 2021-06-01:
> * I just tested with the script provided in our repo, set the protection voltage 3700 mV, the battery cut off smoothly.
> * Then change the protection to 3500 mV (readback to check) and run `stress -c [...]

I understand, but my problem is that none of the registers seem to get modified. Have tried manually using i2cset, also tried modifying protection voltage on upsplus.py and finally tried using this script: 
```
# set_Vfull_Vempty.py

import smbus2

DEVICE_BUS = 1
DEVICE_ADDR = 0x17
SAMPLE_TIME = 1

# new values for Full Voltage and Empty Voltage parameters
FULLVOLTAGE = 4200
EMPTYVOLTAGE = 3700

bus = smbus2.SMBus(DEVICE_BUS)

# enable writing to Full Voltage and Empty Voltage registers
bus.write_byte_data(DEVICE_ADDR, 42,1)

# write to the now enabled registers for Full Voltage and Empty Voltage
print("Set Full Voltage register to: %d" % FULLVOLTAGE)
bus.write_word_data(DEVICE_ADDR, 13, FULLVOLTAGE)
print("Set Empty Voltage register to: %d" % EMPTYVOLTAGE)
bus.write_word_data(DEVICE_ADDR, 15, EMPTYVOLTAGE)

# check result
print("FixedVbat register is: %d" %  bus.read_byte_data(DEVICE_ADDR, 42))

print("Full Voltage register is: %d" % bus.read_word_data(DEVICE_ADDR, 13))
print("Empty Voltage register is: %d" % bus.read_word_data(DEVICE_ADDR, 15))

```
Nothing seems to change registers going from 0x0D to 0x12, I am using version 8 of the firmware

closed 2021-06-10 by nickfox-taterli

leandroalbero, 2021-06-10:
Got an unanswered question :(

tokp, 2021-06-15:
> I understand, but my problem is that none of the registers seem to get modified. Have tried manually using i2cset, also tried modifying protection voltage on upsplus.py and finally tried using this [...]

I have the same issue

nickfox-taterli, 2021-06-16:
Please write byte by byte, please follow the example in the example to write two data byte by byte.

-----------------------------------

ArjenR49, 2021-06-16:
I have the same problem and I never used word write nor read. Always 
taking the value to be written apart in LSB and MSB and then write the 
bytes one by one. LSB first to lower address.

I have a similar setting program for protection voltage and sampling 
interval. One works, the other not. I forgot which, since so much time 
has passed and no new f/w version has come. I think it was the voltage 
setting that didn't work. Fortunately I have a sensible values there 
already.

I am not using the UPS Plus now. It's just sitting there under my Pi 
doing ***@***.*** crunching. Not UPS scripts installed at all now. The 
UPS with v.8 as it is now cannot protect against power failure because 
the scripts cannot write to the i2c registers.

With v.7 we didn't have this problem.

tokp, 2021-06-22:
> Please write byte by byte, please follow the example in the example to write two data byte by byte.
> […](#)

I have update my script to use byte by byte and still have same issues. A sample of my script:
import smbus2

DEVICE_BUS = 1
DEVICE_ADDR = 0x17
SAMPLE_TIME = 1

# new values for Full Voltage and Empty Voltage parameters
FULLVOLTAGE = 4200
EMPTYVOLTAGE = 3300

bus = smbus2.SMBus(DEVICE_BUS)

# enable writing to Full Voltage and Empty Voltage registers
bus.write_byte_data(DEVICE_ADDR, 42,1)

# write to the now enabled registers for Full Voltage and Empty Voltage
print("Set Full Voltage register to: %d" % FULLVOLTAGE)
bus.write_byte_data(DEVICE_ADDR, 13, FULLVOLTAGE & 0xFF)
bus.write_byte_data(DEVICE_ADDR, 14, (FULLVOLTAGE >> 8)& 0xFF)
print("Set Empty Voltage register to: %d" % EMPTYVOLTAGE)
bus.write_word_data(DEVICE_ADDR, 15, EMPTYVOLTAGE)

# check result
print("FixedVbat register is: %d" %  bus.read_byte_data(DEVICE_ADDR, 42))

print("Full Voltage register is: %d" % bus.read_word_data(DEVICE_ADDR, 13))
print("Empty Voltage register is: %d" % bus.read_word_data(DEVICE_ADDR, 15))

mmaga, 2021-12-10:
> > Please write byte by byte, please follow the example in the example to write two data byte by byte.
> > […](#)
> 
> I have update my script to use byte by byte and still have same issues. A sample of my script:
> 
> ```
> import smbus2
> 
> DEVICE_BUS = 1
> DEVICE_ADDR = 0x17
> FULLVOLTAGE = 4200
> EMPTYVOLTAGE = 3300
> 
> bus = smbus2.SMBus(DEVICE_BUS)
> 
> bus.write_byte_data(DEVICE_ADDR, 42,1)
> 
> print("Set Full Voltage register to: %d" % FULLVOLTAGE)
> bus.write_byte_data(DEVICE_ADDR, 13, FULLVOLTAGE & 0xFF)
> bus.write_byte_data(DEVICE_ADDR, 14, (FULLVOLTAGE >> 8)& 0xFF)
> print("Set Empty Voltage register to: %d" % EMPTYVOLTAGE)
> bus.write_word_data(DEVICE_ADDR, 15, EMPTYVOLTAGE)
> 
> print("FixedVbat register is: %d" %  bus.read_byte_data(DEVICE_ADDR, 42))
> 
> print("Full Voltage register is: %d" % bus.read_word_data(DEVICE_ADDR, 13))
> print("Empty Voltage register is: %d" % bus.read_word_data(DEVICE_ADDR, 15))
> ```

@tokp, @ArjenR49 , @leandroalbero, @deHarro 

I managed this script to work adding time.sleep(1) after each bus.write_byte_data() :-)

Sleeping time worked on avoiding Remote I/O error (#121), even when reading a long sequence of bytes from the I2C interface.

The fixes are:

import time
...
# enable writing to Full Voltage and Empty Voltage registers: 42 = 0x2A
bus.write_byte_data(DEVICE_ADDR, 42, 1) 
time.sleep(1)
# write to the now enabled registers for Full Voltage and Empty Voltage
bus.write_byte_data(DEVICE_ADDR, 13,FULLVOLTAGE & 0xFF)
time.sleep(1)
bus.write_byte_data(DEVICE_ADDR, 14,(FULLVOLTAGE >> 8)& 0xFF)
time.sleep(1)
bus.write_byte_data(DEVICE_ADDR, 15,EMPTYVOLTAGE & 0xFF)
time.sleep(1)
bus.write_byte_data(DEVICE_ADDR, 16,(EMPTYVOLTAGE >> 8)& 0xFF)
time.sleep(1)
...

Cheers!