geeekpi/upsplus #18: Some problem on batteries?

hellresistor, 2021-05-10:
Hello
 I have a little test today. I have been used the UPS more than 48hours (with a Fast Charger).
 
 Now I have Disconnected IT. And start get I think to much dropdown.
 
```
  $ python3 upsplusv5-battery-logger.py
['Time (s)', 'Volts (mV)', 'Power (mW)', 'Remaining %']
['84716', '4184', '5463', '99']
['84721', '4198', '4733', '99']
['84726', '4185', '5087', '99']
['84731', '4174', '4686', '78']    ===> When I Disconnect UPS From Charging.
['84736', '4193', '4646', '78']
['84741', '4175', '4673', '78']
['84747', '4200', '4663', '78']
['84752', '4193', '4750', '76']
['84757', '4179', '4646', '76']
['84762', '4190', '4717', '76']
['84767', '4190', '4626', '76']
['84772', '4166', '4669', '76']
['84778', '4169', '4690', '76']
['84783', '4159', '4801', '74']
['84788', '4188', '4851', '74']
['84793', '4175', '4851', '74']
['84798', '4174', '5362', '74']
['84803', '4169', '4885', '74']
['84809', '4180', '4717', '74']
['84814', '4184', '4690', '74']
['84819', '4159', '4690', '74']
['84824', '4166', '4888', '72']
['84829', '4169', '4861', '72']
['84834', '4174', '4834', '72']
['84840', '4172', '4824', '72']
['84845', '4171', '4653', '72']
['84850', '4146', '4619', '72']
['84855', '4167', '4646', '72']
['84860', '4175', '4626', '72']
['84865', '4177', '4636', '72']
['84870', '4167', '4653', '72']
['84876', '4174', '4653', '72']
['84881', '4177', '4669', '72']
['84886', '4134', '4669', '72']
['84891', '4155', '4646', '72']
['84896', '4146', '4619', '72']
['84901', '4158', '5120', '72']  ==> Seems stablised on this value
```

Keep in touch :)

nickfox-taterli, 2021-05-10:
It seems that the battery has a poor discharge capacity (i.e. the voltage is severely pulled down after heavy[?] load), or may be the battery charge/discharge paratemeter has not been learn correctly(by UPS microcontroller), please contact the service and provide the UID to the service for a more detailed check.

hellresistor, 2021-05-10:
Can I say the Blue Led Status keep the 4 lights ON 
and getting on upsLogger
```
['86839', '4088', '4696', '57']
['86844', '4080', '4878', '57']
['86850', '4097', '4656', '57']
```
I Think, should be at least 1 Led OFF ?!

yoyojacky, 2021-05-11:
> root@raspberrypi:~/upsplus# echo ds1307 0x68 > /sys/class/i2c-adapter/i2c-1/new_device
> bash: echo: write error: Device or resource busy
> 
> it seems like you have already let kernel take control on your RTC module, could you please check if there is a parameter in your config.txt file ? like dtoverlay=ds1307,addr=0x48 ?

and please try to comment this line and reboot your raspberry pi and try again.

hellresistor, 2021-05-11:
> root@raspberrypi:~/upsplus# echo ds1307 0x68 > /sys/class/i2c-adapter/i2c-1/new_device
> bash: echo: write error: Device or resource busy
> it seems like you have already let kernel take control on your [...]

Yes I have. Understand that its a 'manual command'.

> 
> and please try to comment this line and reboot your raspberry pi and try again.

After Read the other Issue ticket (you have clarified me) I will do all steps on new RaspiOS installation (Without Batteries). 

Thank you on advance!!

hellresistor, 2021-05-12:
OK. Done!! Thank You!!

closed 2021-05-12 by hellresistor

hellresistor, 2021-05-14:
Thank you to update the Wiki!