geeekpi/upsplus #19: issue running the update

Zbrooklyn, 2021-05-10:
when i run these instructions this is my response

> Method 1
> NOTE: Do not power off or disconnect the network during the upgrade process. If the upgrade fails, the UPS Pro will not work normally.
> 
> Power off the Raspberry Pi.
> Cut off the external charging power (MicroUSB and USB-C)
> Take out all the batteries.
> Press and hold the UPS Pro switch key and insert the battery into the battery compartment.
> At this time, the device will be forced to enter OTA mode.
> The function of the switch button will no longer be available at this time.
> The Raspberry Pi will be running and please execute the following python script in the system terminal to complete the upgrade.
> 
> Download python scripts from github:
> cd ~
> git clone https://github.com/geeekpi/upsplus.git
> cd ~/upsplus
> python3 OTA_firmware_upgrade.py
> The first time it runs, it may prompt that the device is not registered.If it is a legal device, just wait a few seconds and try again.
> 
> UPS Pro Will be turned off after upgrading, Please unplug the power supply, remove the batteries from UPS Pro.
> Insert the batteries back to UPS Pro and then connect power supply and turn it on by press power switch.
> 

```
pi@raspberrypi:~ $ cd ~
pi@raspberrypi:~ $ git clone https://github.com/geeekpi/upsplus.git
fatal: destination path 'upsplus' already exists and is not an empty directory.
pi@raspberrypi:~ $ cd upsplus/
pi@raspberrypi:~/upsplus $ python3 OTA_firmware_upgrade.py
Traceback (most recent call last):
  File "OTA_firmware_upgrade.py", line 18, in <module>
    bus = smbus2.SMBus(DEVICE_BUS)
  File "/home/pi/.local/lib/python3.7/site-packages/smbus2/smbus2.py", line 280, in __init__
    self.open(bus)
  File "/home/pi/.local/lib/python3.7/site-packages/smbus2/smbus2.py", line 310, in open
    self.fd = os.open(filepath, os.O_RDWR)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/i2c-1'
pi@raspberrypi:~/upsplus $ 
```

frtz13, 2021-05-10:
did you activate loading of the i2c kernel module in raspi-config ?
once done, check communication with Full-featured-demo-code.py script

Zbrooklyn, 2021-05-11:
also when i run the first step

i also get this error  
   ```
FAIL] UPS Plus Installation is Incomplete! ... failed!
[info] Please visit wiki for more information:.
```

```
pi@raspberrypi:~ $ curl -Lso- https://git.io/JLygb | bash 
[info] Welcome to 52Pi Technology UPS Plus auto-install Program!.
[info] More information please visit here:.
[info] -----------------------------------------------------.
[info] https://wiki.52pi.com/index.php/UPS_Plus_SKU:_EP-0136.
[info] -----------------------------------------------------.
[info] Start the configuration environment check....
[ ok ] Network status is ok....
[info] Start the software check....
[ ok ] git has been installed..
[ ok ] pi-ina219 library has been installed.
[ ok ] smbus2 library has been installed.
[info] create /home/pi/bin directory....
[info] Create python3 script in location: /home/pi/bin/upsPlus.py Successful.
[ ok ] Create UPS Plus IoT customer service python script successful.
[info] Create crontab list for pi user..
* * * * * /usr/bin/python3 /home/pi/bin/upsPlus.py
* * * * * /usr/bin/python3 /home/pi/bin/upsPlus_iot.py
[info] crontab has been created successful!.
Traceback (most recent call last):
  File "/home/pi/bin/upsPlus.py", line 21, in <module>
    ina = INA219(0.00725, address=0x40)
  File "/home/pi/.local/lib/python3.7/site-packages/ina219.py", line 116, in __init__
    self._i2c = I2C.get_i2c_device(address=address, busnum=busnum)
  File "/home/pi/.local/lib/python3.7/site-packages/Adafruit_GPIO/I2C.py", line 64, in get_i2c_device
    return Device(address, busnum, i2c_interface, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/Adafruit_GPIO/I2C.py", line 97, in __init__
    self._bus = Adafruit_PureIO.smbus.SMBus(busnum)
  File "/home/pi/.local/lib/python3.7/site-packages/Adafruit_PureIO/smbus.py", line 125, in __init__
    self.open(bus)
  File "/home/pi/.local/lib/python3.7/site-packages/Adafruit_PureIO/smbus.py", line 150, in open
    self._device = open("/dev/i2c-{0}".format(bus), "r+b", buffering=0)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/i2c-1'
[FAIL] UPS Plus Installation is Incomplete! ... failed!
[info] Please visit wiki for more information:.
[info] -----------------------------------------------------.
[info] https://wiki.52pi.com/index.php/UPS_Plus_SKU:_EP-0136.
[info] -----------------------------------------------------.
pi@raspberrypi:~ $ 
```

Zbrooklyn, 2021-05-11:
> did you activate loading of the i2c kernel module in raspi-config ?
> once done, check communication with Full-featured-demo-code.py script

i  this  in config
dtparam=i2c_arm=on

```
# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on
```

Zbrooklyn, 2021-05-11:
I'm running it in raspberry pi 4

yoyojacky, 2021-05-11:
> try OTA firmware.. I am getting this issue , very similiar. :|
> 
> ```
> pi@raspberrypi:~ $ cd ~
> pi@raspberrypi:~ $ git clone https://github.com/geeekpi/upsplus.git
> /upsplus $ python3 OTA_firmware_upgrade.py
> Traceback (most recent call last):
>   File "OTA_firmware_upgrade.py", line 22, in <module>
>     aReceiveBuf.append(bus.read_byte_data(DEVICE_ADDR, i))
>   File "/home/pi/.local/lib/python3.7/site-packages/smbus2/smbus2.py", line 433, in read_byte_data
>     ioctl(self.fd, I2C_SMBUS, msg)
> OSError: [Errno 121] Remote I/O error
> ```
> 
> my /boot/config.txt file
> 
> ```
> dtoverlay=i2c-rtc,ds1307,addr=0x68
> dtparam=i2c_arm=on
> [pi4]
> dtoverlay=vc4-fkms-v3d
> max_framebuffers=2
> [all]
> ```

did you assemble it as wiki's instructions ?  and please try to execute this command : i2cdetect -y 1  and paste the results here.

hellresistor, 2021-05-14:
@yoyojacky @nickfox-taterli 
Work to me it was a problem/defect on UPS power button. I am not entering on OTA moda.
🥇

nickfox-taterli, 2021-05-15:
> @yoyojacky @nickfox-taterli
> Work to me finnaly!
> It was a problem/defect on UPS power button. I was not entering on OTA moda.
> 🥇

closed 2021-05-15 by nickfox-taterli

Zbrooklyn, 2021-05-27:
> thank you @nickfox-taterli and @frtz13 for your help finally after I uncommented the i2c I got it working
> 
> I think in the wiki it should say that it is required as step 1
> 
> although I do have 2 questions
> 
> 1. is there a way to safely power down the raspberry pi using the button, right now it justs cuts the power when I push the button
> 2. raspian os has a built-in battery monitor is there a way to show the battery level in the taskbar

does anyone have an answer for this post?