geeekpi/upsplus #21: [Information request] Battery Full Voltage, Empty Voltage, Protection Voltage

frtz13, 2021-05-11:
Who sets or is supposed to set these parameters ?
How are these parameters used by the firmware ?

yoyojacky, 2021-05-11:
please refer to  https://wiki.52pi.com/index.php/UPS_Plus_SKU:_EP-0136

frtz13, 2021-05-11:
sorry, but I cannot find any response to my question on this page (besides the i2c addresses for theses parameters). Can you please be more precise ?

nickfox-taterli, 2021-05-12:
Battery Full Voltage => The maximum voltage to which the battery can be charged is checked, usually to identify the battery type, which is automatic.
Empty Voltage => Checking the minimum voltage to which the battery can be discharged, usually to identify the battery type, is automatic.
Protection Voltage => Once this voltage is reached, the Raspberry Pi will refuse further power, but will continue to be powered when an external power supply is present.

We plan to allow the user to adjust these parameters in a future firmware, making it better than hardware self-learning.

frtz13, 2021-05-12:
so, let's try to understand...
with current firmware release (version 7):
**Battery full voltage**: set by the firmware and not user configurable. is this correct ?
**Empty voltage**: set by the firmware and not user configurable. is this correct?
I do not understand what the firmware uses this value for. remaining capacity calculation?
on my board, empty voltage = 3700mV.
is there a calibration procedure to make "automatic detection of battery type" work properly? there should also be a procedure how to reset automatically detected parameters when replacing batteries.
**protection voltage**: set by user (it is set by the installation procedure to 3700mV). User configurable. When battery voltage < protection voltage and battery not charging, the firmware will shut down output power ?

nickfox-taterli, 2021-05-13:
The firmware sampling voltage is used for power, charging and discharging capacity estimation, if you edited manual way to specify him, we will trust your setting, the role of protection voltage as you said.

closed 2021-05-15 by nickfox-taterli