geeekpi/upsplus #23: Raspberry Shutdown but not shutdown

hellresistor, 2021-05-13:
Hello
Have executed `sudo shutdown now` on raspberry pi to halt the rpi.

But seems after turn off rpi leds like 1 second turn red.

Tried to shutdown by UPS buton. but not happens.

Some way to force shutdown ?

frtz13, 2021-05-13:
which version of the firmware are you using? be sure to use version >= 7. otherwise, you should upgrade the firmware.
did you start a shutdown countdown for the UPS before shutting down the RPi ? i2cset -y 0x17 24 30

hellresistor, 2021-05-14:
> which version of the firmware are you using? be sure to use version >= 7. otherwise, you should upgrade the firmware.
> did you start a shutdown countdown for the UPS before shutting down the RPi ? i2cset -y 0x17 24 30

will recheck that. Ty!

PS: I am a newbie with this sorry this boring questions.
PS2: I have seen wiki updated :)

nickfox-taterli, 2021-05-14:
You have to inform the UPS that you are going to shut down, otherwise he will not shut down.

Of course, you can make it a shutdown-triggered task (systemd) if you need to.

hellresistor, 2021-05-14:
I have find the issue about "pressing the button" and do nothing.
One of my UPS's arrived with a little defectiv (not Opened the other piece yet, testing this one first). the button is not nice solder on 'k1' . the button mooves to back.
Solution. Done a Upper pressure to downand than click the button, and OK!
 About this little thing. Can I put some 'hot Glue' between button and batterie support, this will Fix/stop button moving/shaking, I am authorized do that?

nickfox-taterli, 2021-05-14:
If it is a product quality problem, it is recommended to contact the after-sales service, because here can only deal with technical problems.

hellresistor, 2021-05-14:
> which version of the firmware are you using? be sure to use version >= 7. otherwise, you should upgrade the firmware.
> did you start a shutdown countdown for the UPS before shutting down the RPi ? i2cset -y 0x17 24 30

Firmware Upgrade DONE! 
I have set To 10 with this command.  `i2cset -y 1 0x17 24 10` (10 seconds).
than not work and set UP to 30. and now. shutdown like 3 seconds after OS off!

Should be Seconds ou Miliseconds on the `i2cset -y 1 0x17 24 30` ?

I will set close! Thank you all!!

about your last message @nickfox-taterli have some contact or official place to send mail?

hellresistor, 2021-05-14:
> You have to inform the UPS that you are going to shut down, otherwise he will not shut down.
> 
> Of course, you can make it a shutdown-triggered task (systemd) if you need to.

I presume it is this ?
i2cset -y 1 0x17 25 1
i2cset -y 1 0x17 24 30

hellresistor, 2021-05-14:
Heya again.

I build this steps to lazy ^^

```
sudo su
apt update && apt upgrade -y
### ENABLE I2C com sudo rasp-config -> Interfaces -> Enable I2C 
sudo pip install smbus2 
sudo apt install -y i2c-tools
sudo modprobe i2c-bcm2835
sudo modprobe rtc-ds1307
sudo su 
echo ds1307 0x68 > /sys/class/i2c-adapter/i2c-1/new_device
sed -i '/^dtparam=i2c_arm=on/a dtoverlay=i2c-rtc,ds1307,addr=0x68' /boot/config.txt 
apt remove -y fake-hwclock
update-rc.d -f fake-hwclock remove
sed -i '/if \[ -e \/run\/systemd\/system \] \; then/,+2 s/^/#/' /lib/udev/hwclock-set
sudo hwclock -w 
sudo hwclock 
reboot
cd ~
# Edit your configs all 'SAMPLE_TIME' variables
curl -Lso- https://git.io/JLygb | bash

# Back-On AC detect
i2cset -y 1 0x17 25 1
# Shutdown countdown
i2cset -y 1 0x17 24 10    <=== HERE should be 0x18 10 instead 0x17 ?!
```

I am not understanding what is the 24 and 25 values? the 0x17 understand it is Power Status/Operation Mode?

With this configuration I find this situation:
 With Power Cable Connected on UPS and RaspiOS running and configured with steps I have metionated before 
  - Send command `sudo shutdown` or `sudo halt`
  - rpi start shutdown process, shutdown screen (have a fan, keeps turned on, but this it is becaus it was connected on GPIO2and3 (+3.3v and gnd) and normal, receive energy direct from UPS 
  - RPI and UPS keeps Red Led ON ( after The RPI red Led blink). 
  - Click on UPS Power button, do nothing, keeps both Red Light ON.
  - Disconnect Power cable from UPS, keep the Red Lights On
  - Click on UPS Power button again, AND YES. Full PowerOff the RPI and UPS (Red Leds OFF).

Will should be possible power off the rpi/ups from UPS button, with and without power cable?

Thank you to all help

hellresistor, 2021-05-14:
> Thank you to all help
> 
> Actually i am testing if shutdown securely rpi4 when battery low

well I used the upsloggerv5 (seems not correct version because of values return).  and when reach the 1% the OS shutdown auto but keeps the double Red Led on (nor poer off). The Blue leds have got 2 ON fix yet, means have 25%~50% battery.. And power button ups work (yes, not power cable connected). I will replace with another new pair of batteries (3rd round and last)..

Hope get some lights

nickfox-taterli, 2021-05-15:
> well I used the upsloggerv5 (seems not correct version because of values return). and when reach the 1% the OS shutdown auto but keeps the double Red Led on (nor poer off). The Blue leds have got 2 [...]

When lower than the [Protection Voltage] voltage, will immediately shut down the output, you can adjust the [Protection Voltage] to a smaller value, as far as I know there are batteries can support the work of 2400mV, why the default will be 3700, because some batteries in the late discharge heat will be more serious.

Why press the power button available? because once the load is withdrawn, the battery voltage will be higher than the protection voltage, that is, you use the battery with limited load capacity, so there is a load can take below the protection voltage, if you think the protection voltage mistake,it can be set to lower, you can artificially adjust.

> Heya again.
> 
> I build this steps to lazy ^^
> 
> ```
> sudo su
> apt update && apt upgrade -y
> ### ENABLE I2C com sudo rasp-config -> Interfaces -> Enable I2C 
> sudo pip install smbus2 
> sudo apt install -y i2c-tools
> sudo modprobe i2c-bcm2835
> sudo modprobe rtc-ds1307
> sudo su 
> echo ds1307 0x68 > /sys/class/i2c-adapter/i2c-1/new_device
> sed -i '/^dtparam=i2c_arm=on/a dtoverlay=i2c-rtc,ds1307,addr=0x68' /boot/config.txt 
> apt remove -y fake-hwclock
> update-rc.d -f fake-hwclock remove
> sed -i '/if \[ -e \/run\/systemd\/system \] \; then/,+2 s/^/#/' /lib/udev/hwclock-set
> sudo hwclock -w 
> sudo date && sudo hwclock -r
> reboot
> 
> cd ~
> # Edit your configs all 'SAMPLE_TIME' variables
> curl -Lso- https://git.io/JLygb | bash
> 
> # Back-On AC detect
> i2cset -y 1 0x17 25 1
> # Shutdown countdown
> i2cset -y 1 0x17 24 10    <=== HERE should be 0x18 10 instead 0x17 ?!
> ```
> 
> I am not understanding what is the 24 and 25 values? the 0x17 understand it is Power Status/Operation Mode?
> 
> With this configuration I find this situation:
> With Power Cable Connected on UPS and RaspiOS running and configured with steps I have metionated before
> 
> * Send command `sudo shutdown` or `sudo halt`
> * rpi start shutdown process, shutdown screen (have a fan, keeps turned on, but this it is becaus it was connected on GPIO2and3 (+3.3v and gnd) and normal, receive energy direct from UPS
> * RPI and UPS keeps Red Led ON ( after The RPI red Led blink).
> * Click on UPS Power button, do nothing, keeps both Red Light ON.
> * Disconnect Power cable from UPS, keep the Red Lights On
> * Click on UPS Power button again, AND YES. Full PowerOff the RPI and UPS (Red Leds OFF).
> 
> Will should be possible power off the rpi/ups from UPS button, with and without power cable connected on UPS?
> 
> Thank you to all help
> 
> Actually i am testing if shutdown securely rpi4 when battery low

The current design is short press to turn on, long press to turn off, if you set Countdown, you can automatically turn off, but according to the Wiki reference, you can only set Countdown for 10 seconds or more.

hellresistor, 2021-05-15:
Should be normal show the `UU` instead 68 ?

```
pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- 17 -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: 40 -- -- -- -- 45 -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

```

nickfox-taterli, 2021-05-15:
> Should be normal show the `UU` instead 68 ?
> 
> ```
> pi@raspberrypi:~ $ i2cdetect -y 1
>      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
> 00:          -- -- -- -- -- -- -- -- -- -- -- -- --
> 10: -- -- -- -- -- -- -- 17 -- -- -- -- -- -- -- --
> 20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 40: 40 -- -- -- -- 45 -- -- -- -- -- -- -- -- -- --
> 50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
> 70: -- -- -- -- -- -- -- --
> ```
> 
> `i2cset -y 1 0x17 24 60 ` when i run this command. after that time (60sec) , the rpi/ups do the poweroff (if power cable DISConnected) without executing shutdown command on terminal...

The correct order is to set the countdown timer and then execute the normal system power off, because the two are independent, please refer to our [script](https://github.com/geeekpi/upsplus/blob/main/upsplus.py#L94
) ,68h is the RTC device, normally you just read the system time, if you must access through the direct programming method, please delete the RTC dtoverlay in config.txt.

closed 2021-05-16 by hellresistor