geeekpi/upsplus #24: scheduled power cycling using a script

ArjenR49, 2021-05-13:
I use the script below for a scheduled power cycle of the Pi attached to the UPS.
I run it by cron every 15 minutes to stress test the UPS.
Mostly the script will indeed shut the Pi down, then after the designated time, cut the power to the Pi, and then after 5 seconds more, power the Pi on again -> a full power cycle.

PROBLEM: Every now and then, usually when I am not looking, it fails to cut the power to the Pi. The Pi is then just sitting there waiting for me to press the button (twice) to get it going.

My intention, of course, is to be able to power cycle the Pi at an appropriate time, say one per 24 h, to safeguard against software that performs poorly in some way.

PowerCycle.py:

#!/usr/bin/python3
# -*- coding: utf-8 -*-

# adapted from scripts provided at GitHub: Geeekpi/upsplus by nickfox-taterli
# ar - 13-05-2021

# ''' Halt the Pi, power down, then power up Pi (= perform power cycle) '''

import os
import time
from smbus2 import SMBus 

# Define I2C bus
DEVICE_BUS = 1

# Define device I2C slave address.
DEVICE_ADDR = 0x17

# Essential UPS I2C register default values
# (name format: Operation Mode Register Address)
# for shut down & power off state
OMR0x18 = 0   # seconds, power off delay
OMR0x19 = 0   # boolean, automatic restart or not
OMR0x1A = 75  # seconds, power on delay

# Write byte to specified I2C register address
def putByte(RA, byte):
    while True:
        try:
            with SMBus(DEVICE_BUS) as bus:
                bus.write_byte_data(DEVICE_ADDR, RA, byte)
                break
        except TimeoutError:
            time.sleep(0.1)

print("*"*62)
print(("*** {:^54s} ***").
      format("Make the UPS sync & shut down Pi in an orderly manner"))
print(("*** {:^54s} ***").format("without delay & turn power off after "
      + str(OMR0x1A)+" seconds."))
print(("*** {:^54s} ***").
      format("Five seconds later the UPS will power up the Pi again."))
print("*"*62)
print()
print()

# Disable/enable automatic restart on return of external power.
#   Enable: write 1 to register 0x19
#   Disable: write 0 to register 0x19
putByte(0x19, OMR0x19)
time.sleep(2)

# For power cycle operation only 'power up' countdown (0x1A) must be set.
# UPS will cut power to the Pi 5 seconds before the end of the power up
# countdown, allowing the Pi to sync & halt in an orderly manner,
# and then turn power to the Pi on again at 0 seconds.
putByte(0x1A, OMR0x1A)
time.sleep(2)

# Halt the Pi without delay.
os.system("sudo shutdown now")

# Script continues executing, indefinitely as it were (& keeping the lock)
# until it is eventually killed by the Pi shutting down.
while True:
    time.sleep(10)

# Control is now left to the UPS' F/W and MCU ...
# EOF

ArjenR49, 2021-05-13:
Sorry about how that looks now. I didn't expect that to happen. Newbie in GitHub ...

hellresistor, 2021-05-13:
> Sorry about how that looks now. I didn't expect that to happen. Newbie in GitHub ...

Select all 'code' and Than click on < > icon ;) (because of multiline code) Edit it  ;)

ArjenR49, 2021-05-14:
> > Sorry about how that looks now. I didn't expect that to happen. Newbie in GitHub ...
> 
> Select all 'code' and Than click on < > icon ;) (because of multiline code) Edit it ;)

Thanks. That was an improvement. However ....

ArjenR49, 2021-05-14:
Upgraded to f/w 7 by method 1.
Unfortunately not an improvement here.

When I now run my powercycle.py script (see above), the Pi never starts up again. The UPS cuts off power to the Pi at the end of the countdown, but then .... nothing happens.
I have to take out the batteries. On f/w v.5 this used to work fine (most of the time). After some seconds the Pi would get power again and start up.
What my powercycle script does is simply write 0 to off-set 0x19 and 75 to off-set 0x1A (off-set 0x18 not written to, it remains at 0) and issue a 'sudo shutdown now' command to the Pi.

ArjenR49, 2021-05-14:
As a side effect of the long time till startup I notice that charging actually finishes, i.e. all four blue LEDs are on. None blinking.

nickfox-taterli, 2021-05-14:
We do not recommend that you use this function to reboot, because this is an plug-in and plug-out power (means hard reset), if you have to system has been frozen, it may not be able to send this command, but we have a solution, you can try to set the reboot timer 120 seconds per minute, so long as the implementation of this command every minute, it will not lead to reboot, frequent reboots may fail, this is because the battery charging rate is much lower than the discharge rate, and completely shut down until the full power on this process is not charging, this is for protection reasons, because this time there is no perfect temperature/charge rate control measures.

We recommend that you write the test procedure as a list, so that we can follow the list of steps to test.

1. step 1
2. step 2
3. step 3

nickfox-taterli, 2021-05-14:
> As a side effect of the long time till startup I notice that charging actually finishes, i.e. all four blue LEDs are on. None blinking.

Charging will be completed if your battery capacity is small (usually because the parameters marked on the battery are exaggerated), but he will recheck the battery status at every interval (adjustable, default is 2 minutes.) Re-check the battery status, which will re-trigger the charging function.

ArjenR49, 2021-05-14:
Thank you nickfox-taterli, for your answers.
There is no problem with the charging. It was just an observation. Some people 'complained' that charging never seems to finish ... 
I have made the observation that when the UPS is doing its own checking and extinguishes the blue LEDs for a little while, the batteries are used for powering the Pi and will be shown to discharge at that moment. When the lights come back on, the batteries are charged again. So, batteries continually get discharged a bit and then charged again. No problem for me. My batteries are made by sanyo and from a reliable local supplier.
I don't run down the batteries during my testing.
I use my own upsPlus.py modified from yours. It has a time counter of its own, which counts down the minutes the UPS has been without external power and when that 'grace time' is up, it executes the same code that upsPlus executes when the batteries are low because of no external power.
I do this so, because this way I don't have to wait until batteries are depleted to test the 'restart when AC returns'.
A second reason is to imitate the 'grace time' regime on a UPS made by Olmatic that I have: 'stop the Pi when power is gone for 10 minutes', and restart when it comes back on ....

ArjenR49, 2021-05-14:
You are totally right that when the Pi freezes, the upsPlus.py control program cannot do much. It probably won't even run on the minute in that situation. 

I do not currently really expect this Pi to freeze, but I like to do 'keep alive' by scheduled reboot on some Pi's every day. Same for the router (not a Pi) at my cottage, because there's often nobody available to reset anything manually.

I could just schedule a simple 'sudo reboot now'. It is one of the options to consider. I also often install the watchdog built into the Pi. It uses the GPU, if I remember correctly. I have tested it with a 'forkbomb'. It works.

I hesitate to say that I found it hard to grasp what you mean further down in the comment which you started with 'we do not recommend that you use this function to reboot'. Could you please rephrase it? It looks a useful piece of documentation.

ArjenR49, 2021-05-14:
If anybody is interested, my scripts can be found here: https://github.com/ArjenR49/UPS-Plus
I am new to GitHub and have only little experience withpython. I did a lot of rewriting the scripts and formatting all output to my liking. Learned a lot.

nickfox-taterli, 2021-05-15:
> You are totally right that when the Pi freezes, the upsPlus.py control program cannot do much. It probably won't even run on the minute in that situation.
>
> I do not currently really expect this Pi [...]

Turn off all the lights and turn on at intervals, that parameter is called [Sample Period], this is because some batteries will appear false full phenomenon, as the battery no-load 4.2V, with a load immediately less than 3.7V, if you ensure that your battery is good, you can properly adjust this parameter

> If anybody is interested, my scripts can be found here: https://github.com/ArjenR49/UPS-Plus
> I am new to GitHub and have only little experience with python. I did a lot of rewriting the scripts and formatting all output to my liking. Learned a lot and managed to not break anything ;-) I think.

Because the use of UPSv5 for reboot, actually simulates the plugging and unplugging of external power supplies, which may damage your file system, which is useful when the system is unresponsive, you can also impl. this is a watchdog function, when did not feed the reboot timer, will lead to the end of the countdown to force a reboot.

> If anybody is interested, my scripts can be found here: https://github.com/ArjenR49/UPS-Plus
> I am new to GitHub and have only little experience with python. I did a lot of rewriting the scripts and formatting all output to my liking. Learned a lot and managed to not break anything ;-) I think.

We are interested in your project, it would be great if you have a UI, please write us an email if you want, if we have some internal test firmware for small scale testing, we can also provide you with it.

E-Mail: @yoyojacky <jacky.li@52pi.com>

ArjenR49, 2021-05-16:
PowerCycle.py may not be useful when the Pi freezes - thankfully we have several watchdogs - but it doesn't damage the file system as the script shuts down the Pi well ahead of the UPS cutting the power to the Pi.
If one uses the countdown timer for restarting, under f/w v.5 the Pi used to be started up 5 seconds after the power was cut, but as from v.7 this lasts some 10 minutes.
Maybe there is room for choice by the user; cutting the 10 min short with a button press would be great.

nickfox-taterli, 2021-05-16:
It is possible and not 100%, as long as it is because the correct shutdown sequence was not executed, which is possible (although rare) to cause file system corruption, and the system has frozen, he will not be able to save any content, only forced power off, which is not executed the correct shutdown sequence, because at that time it is not possible to execute the correct shutdown sequence.

closed 2021-05-16 by ArjenR49