geeekpi/upsplus #25: Restart after shutdown by the OS itself (in the GUI)?

ArjenR49, 2021-05-15:
f/w version 7
upsPlus.py running every minute and setting 0x18-0x1A registers to 0, 0, 180 respectively (watchdog function).
The did a showdown from the GUI (in Raspberry OS).
Pi shuts down. After some time, power is cut by the UPS. Soon batteries become fully charged as seen by 4 blue LEDs, none blinking any longer.
All good ...
But how to restart the Pi?

I tried pressing the UPS button, unplugging the charger, but no reaction from the Pi.

However, eventually it did start again. Ahaa, the watchdog timer must have done that.

So the UPS is not aware of the user shutting down the Pi, and therefore the UPS just goes on doing its watchdog routine and doesn't check if the user is pushing the button.

Can this maybe be changed?
If after a watchdog time-out, the user is present and pushes the button, could the 9 minutes restart delay be cut short and the Pi restart immediately?

I was getting ready to screw off the bottom plate again and remove the batteries as I thought the UPS had locked up somehow ... then a little later it was running again. Good thing I have the colourful 52Pi Ice Tower, it shows from far.
(My Pi's bootloader config is such that the Pi stops the fan when it shuts down.)

Having documentation explaining that would prevent such panic ;-) I realize writing documentation takes time, but eventually such things should be documented.

ArjenR49, 2021-05-15:
I did a new test.
Shut down the Pi from the Raspberry OS GUI.
I didn't touch anything.
After a while the UPS cut the power to the Pi.
After still more time, which seemed longer than 9-10 minutes, the Pi started up again by itself.

So, if you want to shut down the Pi and have it stay off, it's not going to work as it is, with the 'watchdog settings' in upsPlus.py (and therefore in the UPS registers).
How can this be solved? Taking the batteries out is not a very good solution.

nickfox-taterli, 2021-05-15:
1)You can rewrite the countdown, even if he is counting.
2)You are writing [Restart Countdown], of course it will restart, there is another register [Shutdown Countdown] that does not restart.
3)According to the test, the button is not responding during the countdown, we will plan to improve this feature.

ArjenR49, 2021-05-15:
Following your advice, I also tested watchdog timer functionality without restarting (set 0x18 to >=120, 0x19 and 0x1A to 0). It works as promised. Thank you!
Restarting by pressing the button.

closed 2021-05-15 by ArjenR49