geeekpi/upsplus #26: Under-voltage detected

frtz13, 2021-05-15:
sometimes I get "under voltage" warnings in my raspberry pi log.
running Raspberry Pi 4b + SSD + UPSPlus.

in syslog several events like:

20:06:52 rpi kernel: Under-voltage detected! (0x00050005)
20:07:17 rpi kernel: Voltage normalised (0x00000000)

to investigate the issue, I run a background task on the RPi which calls the INA219 sensor at address 0x40 every second.
for each one minute interval it determines the minimum output voltage and the maxium output current and sends it to the graphic.

[image]

this seems to show that the peaks of 2.7A output current make drop the output voltage as measured by the INA219 sensor to below 4.87 V which triggers the under-voltage warnings.
is there a setting to set the output voltage a little bit higher?

nickfox-taterli, 2021-05-16:
short voltage drop does not cause a problem, as the Raspberry Pi itself is designed for 2.5A maximum, if your SSD supports external power supply you can also take power from the USB output of the UPS (but this part will not be measured) to reduce the pressure on the Raspberry Pi.

frtz13, 2021-05-16:
my SSD does not accept an external power supply.
is there a way to get the output voltage to a higher level, say up to 5.0V ? in your tutorial video https://www.youtube.com/watch?v=jQljQJ41I5A I saw that the UPS output voltage measured by the ina219 sensor was even above 5.1V. even at lower current draw, the output voltage of my UPS board never goes up to 5.0V .

nickfox-taterli, 2021-05-18:
This voltage is a hard protection, modify the ratio of resistors R3/R5, the higher their ratio, the higher the output voltage, when the ratio is 3.15, the output is 5V, the maximum can be adjusted to 6V, but 6V may damage the Raspberry Pi.

frtz13, 2021-05-19:
what are the factory built-in values for R3 and R5 ?

nickfox-taterli, 2021-05-28:
R3:220K 1%,R5:69.8K 1%

closed 2021-05-28 by frtz13