geeekpi/upsplus #30: Using the RTC to schedule some event on the UPS?

ArjenR49, 2021-05-17:
In connection with another UPS product adding a RTC to the Pi, also on a UPS board, it was said that the RTC on its own can trigger one single event. I don't know what it requires, but I assume it needs to be enabled by the hardware, too. It was used on that UPS product (made by Olmatic, Germany) to schedule a boot event. I used it for making sure the Pi would always be started up at least once a day at a certain time, in case it had somehow not restarted otherwise.

Are there plans to use the RTC for scheduling something on the UPS Plus?
Does the current hardware even allow it?

nickfox-taterli, 2021-05-18:
You can try set-up a system cron, for example, as follows:

`0 0 * * * i2c-set -y 0x17 0x1A 60`

This will allow you to perform a reboot every night starting at 0:00.

ArjenR49, 2021-05-18:
Sure, but that depends on the Pi running.
The RTC doesn't require the Pi to be running. It could be frozen ...
But then there's the different watchdogs, on the Pi itself and on the UPS.

Maybe your developers can think of a use for the RTC trigger function in a future version.

closed 2021-05-18 by ArjenR49