geeekpi/upsplus #31: What if upsPlus.py consistently fails to run

ArjenR49, 2021-05-17:
or isn't found?
Essentially the question is, what will the f/w do, if there is no control program and external power fails?
Will it eventually stop draining the batteries at a voltage still safe for the batteries?

hellresistor, 2021-05-17:
i think have discovered that tonight. or that, or some bad configuration on `raspi-blacklist.conf` file.
On the morning got my rpi OFF and batteries empty.
:o

ArjenR49, 2021-05-17:
Well, the question is how empty? Below 2,5 V?  Or did you mean the ups did
its job on its own and switched power to the pi off to protect the battery?

Did you mean that I2C was disabled on the Pi?

I had to Google raspi-blacklist.conf and it was suggested raspberry os
doesn't have a blacklist like that. I am only familiar with raspi-config.

Arjen
(On The Road)

frtz13, 2021-05-17:
according to a comment in issue #21, the firmware will shut down the output voltage of the UPS when battery voltage goes below the "protection voltage" (ups registers at address 0x11 / 0x12)

hellresistor, 2021-05-17:
> Well, the question is how empty? Below 2,5 V? Or did you mean the ups did its job on its own and switched power to the pi off to protect the battery? Did you mean that I2C was disabled on the Pi? I [...]

- have reverted-back raspi-blacklist.conf file. 
- batteries (4x) with 4.2v (3.7v) fullcharged.
- Click power button (NO power cable connect), Ups on and detecting battery low (1 led blink), raspberry and OS run.
- Re-run the upsplus.git repository, execute uninstall.sh and then install.sh
- Fast/drastic power off of rpi and ups power
- Connect power cable. UPS/Rpi power on auto. and wait until reach 2nd led blinking (on 1st try will happens error, because script detect low battery and send the  power off)
-  Execute `curl -Lso- https://git.io/JLygb | bash`

Seems working and show charging leds now 1 led fix 2nd led blink
23:23

```
------------------------------------------------------------
------Current information of the detected Raspberry Pi------
------------------------------------------------------------
Raspberry Pi Supply Voltage: 4.908 V
Raspberry Pi Current Current Consumption: 1135.071 mA
Raspberry Pi Current Power Consumption: 5571.068 mW
------------------------------------------------------------
-------------------Batteries information-------------------
------------------------------------------------------------
Voltage of Batteries: 3.764 V
Battery Current (Charging) Rate: 4167.805 mA
Current Battery Power Supplement: 15726.829 mW
Successfully set the protection voltage to: 3700 mV
------------------------------------------------------------
Currently charging via Type C Port.
```

1minutes later
```
------------------------------------------------------------
------Current information of the detected Raspberry Pi------
------------------------------------------------------------
Raspberry Pi Supply Voltage: 4.908 V
Raspberry Pi Current Current Consumption: 1131.034 mA
Raspberry Pi Current Power Consumption: 5793.103 mW
------------------------------------------------------------
-------------------Batteries information-------------------
------------------------------------------------------------
Voltage of Batteries: 3.784 V
Battery Current (Charging) Rate: 4203.902 mA
Current Battery Power Supplement: 15887.805 mW
Successfully set the protection voltage to: 3700 mV
------------------------------------------------------------
Currently charging via Type C Port.
```

+1minutes later

```
------------------------------------------------------------
------Current information of the detected Raspberry Pi------
------------------------------------------------------------
Raspberry Pi Supply Voltage: 4.912 V
Raspberry Pi Current Current Consumption: 1075.862 mA
Raspberry Pi Current Power Consumption: 5281.749 mW
------------------------------------------------------------
-------------------Batteries information-------------------
------------------------------------------------------------
Voltage of Batteries: 3.784 V
Battery Current (Charging) Rate: 4129.756 mA
Current Battery Power Supplement: 15536.585 mW
Successfully set the protection voltage to: 3700 mV
------------------------------------------------------------
Currently charging via Type C Port.
```

ArjenR49, 2021-05-17:
Why did you install the software again?
If an original script has changed, isn't it easier to just download the
newer version?

I rewrote the scripts because I found the output formatting terrible.
Values accurate to the mV, mA and mW are not credible, let alone decimal
fractions of those units.

You wrote you have 4 batteries. Did you make your own expansion board
connecting to the 2 groups if 4 pins?

Arjen
(On The Road)

hellresistor, 2021-05-17:
> You wrote you have 4 batteries. Did you make your own expansion board connecting to the 2 groups if 4 pins?

NONONO!!!!!! i use 2 and have other to for replacement

ArjenR49, 2021-05-18:
With the watchdog  (0x18 being set to 180 in upsPlus every time it runs), and all the exception handling that I can throw at it, it still happens rather often that 0x18 is NOT updated to 180 by upsPlus.py.
At the moment I have taken out most time.sleep().
There is something very strange to me going on with that SMBus writing. Something I can't find.
And so, every morning early I find the Pi stopped. In the UPS_event.log, I see regular cases where my upsPlus.py reports 0x18=120 ... (first it really hard tries to update to 180 .... and then a few lines of code later the script reads 120 from that register .... it should always be either 180 or 179 .... and then when upsPlus is done, the f/w counts down to 120 .... until 0x18 gets reset to 180 by the script running again one minute later.

Next I'll  take out the watchdog .... have the upsPLus.py set 0x18 to 0. And test again ...
And add time.sleep(0.25) or so back in in those places where something bad goes on.

I'm also thinking of making a short version of upsPlus.py with no other output than timestamp and 0x18 content. That way I'll have a log that's short enough to post ...

Also I found that not updating 0x18 correctly is apparently not the reason for PowerCycle.py to fail to restart the Pi.
In v.7 it SHOULD restart the Pi ca. 10 minutes after starting the script (which sets  0x18-0x1A and then shuts down the Pi; the UPS cuts power and then 10 min. later it poweers the Pi on again .... most of the time .... sometimes it fails to start up the Pi ...
in v.5 it wouldn't take 10 minutes, just 5 seconds. They changed it .... probably changed something radical in the f/w ...

yoyojacky, 2021-05-24:
> With the watchdog (0x18 being set to 180 in upsPlus every time it runs), and all the exception handling that I can throw at it, it still happens rather often that 0x18 is NOT updated to 180 by [...]

hi ArgenR49, 
you do not need to always set 0x18 register , you can comment it out , it is Shutdown Countdown setting , it will start countdown once you set it to a number in range 10 ~255 ,
for example, it will told ups to shutdown power after 240 seconds. 
`i2cset -y 1 0x17 24 20`
 it means starting countdown in 20 seconds. 
why i put `24` after `0x17`?
`0x18 | Shutdown Countdown | 0(False)/10 - 255 | Bool/Sec` 
beacuse 0x18 in binary is 00011000, decimal number is : 24
and you can modify your upsPlus.py file to fit for your needs,  it is a demo code for you.

closed 2021-05-24 by ArjenR49