geeekpi/upsplus #34: UPS not charging batteries correctly

hellresistor, 2021-05-19:
Hello friends.
Yesterday have  arrived to home and see my rpi off (with no time), today i have checked... the UPS are not charge the batteries correctly. Actually batteries have ~2.1V each.
I am using a 3Amp quality  charger.

Maybe this issue have about the Fw7 ?

ArjenR49, 2021-05-19:
You could run upsPlus.py and reroute its output to a file. Then you'll have some kind of a trail. You can also run my UPS_report.py and reroute the output to a file. It will supply more information.
I run upsPlus with output to a file all the time. Maybe I will not do that any longer when I leave it to run for months at a time eventually, but for now.
It would be easy to add more output to either script as they both read the whole of the UPS memory into the buffer called aReceiveBuf.

nickfox-taterli, 2021-05-20:
If you would like us to assist you with your troubleshooting, you will need to provide your UID to the [after-sales service], please do not provide it here.

ArjenR49, 2021-05-20:
I need to find that service first, then. I assumed this place is it.
BTW, I have no problem with batteries not being charged properly. That was the other person who started this issue.

hellresistor, 2021-05-20:
> If you would like us to assist you with your troubleshooting, you will need to provide your UID to the [after-sales service], please do not provide it here.

ok. i will do reference on this post.

> I need to find that service first, then. I assumed this place is it.
> BTW, I have no problem with batteries not being charged properly. That was the other person who started this issue.

what is website you have purchased the ups?

ArjenR49, 2021-05-20:
I bought my UPS plus board from the 52Pi Official Store on AliExpress.

I have not had problems with the charging as such.

I have had to solve problems with TimeoutError exceptions and failing writes (to 0x18 & 0x1A, the countdown registers). That was really bad, actually. Cost me a lot of time and effort.

The voltage to the Pi is slightly below 5V, but no problems because of that. (headless, no monitor).

It can very well be that the current readings for the battery are off, as somebody wrote in another issue, but otherwise it all seems to wrks logically. When the bluelights go out, the battery is discharging and the UPS collects data about it. The discharging lasts only a short time (less than 15 sec) and the batteries get charged during the rest of the time.
Whenever the Pi is off, the batteries get fully charged after a short while and all 4 blue leds stay on.

I am now using Frtz13's scripts. It doesn't have all the same features as what I had built into my scripts, but I was only mimicking another UPS (Olmatic) and these extra features aren't essential to me.
After I removed the charger, my Pi ran another hour with the default setting of 3,7V. (This I don't understand; it is waaay higher than what the lithium cells can discharge to safely.)

I think I'll stick to Frtz13's scripts. I can change my own UPS_report.py to fit better with that. It has rounding to significant digits and V, A and W instead of mV, mA and mW.
I suppose I could even use my own PowerCycle.py script if I want. So I only replace the GeeekPi's control program (or my own version of it) with fanShutDownUps.py.
I don't have the hardware for the Ice Tower fan control now, but that's easy (unless the LEDs can't handle the PWM). The MQTT stuff is totally new to me, but interesting since I have a cottage where I have little servers etc.
Frtz's is a complete package and running in the background. Really Nice!

closed 2021-05-22 by hellresistor