geeekpi/upsplus #39: config.txt different directory on distro

hellresistor, 2021-05-22:
Hello
I am using this image [Debian 10 raspberry](https://raspi.debian.net/verified/20210210_raspi_4_buster.img.xz)
I am getting this directory  `/boot/firmware/config.txt` instead the default `/boot/config.txt`

The file contains:
```
# Switch the CPU from ARMv7 into ARMv8 (aarch64) mode
arm_64bit=1

enable_uart=1
upstream_kernel=1

### UPSPro Config Added manually ###
dtoverlay=i2c-rtc,ds1307
dtparam=i2c_arm=on
##################

kernel=vmlinuz-5.10.0-0.bpo.5-arm64
# For details on the initramfs directive, see
# https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=10532
initramfs initrd.img-5.10.0-0.bpo.5-arm64
```

I cannot getting info from UPS  .

Executing the script, getting  this :|
```
$ sudo python3 bin/upsPlus_iot.py
[sudo] password for ha:
Traceback (most recent call last):
  File "bin/upsPlus_iot.py", line 16, in <module>
    ina = INA219(0.00725,address=0x40)
  File "/usr/local/lib/python3.7/dist-packages/ina219.py", line 116, in __init__
    self._i2c = I2C.get_i2c_device(address=address, busnum=busnum)
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_GPIO/I2C.py", line 63, in get_i2c_device
    busnum = get_default_bus()
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_GPIO/I2C.py", line 55, in get_default_bus
    raise RuntimeError('Could not determine default I2C bus for platform.')
RuntimeError: Could not determine default I2C bus for platform.
```

have tried create a link with ln -s /boot/firmware/config.txt /boot/config.txt , but not work.

PS: The RPI leds are off (both green/red)

yoyojacky, 2021-05-24:
yes, you are using 64bit OS, it is a bit different from 32bit OS, and if you can enable I2C on this OS ? because the daughter board is communicate with raspberry pi via I2C protocol via USB-C port , you need to enbale dwc2 function so that it can communicate with each other, and for 32bit, just adding `dtoverlay=dwc2,dr_mode=host` to /boot/config.txt file and reboot raspberry pi is ok. I am not sure about if 64bit OS also support that dtb file as well.

hellresistor, 2021-07-23:
I am getting this output

```
teste@rpi4-20210718:~$ python3 bin/upsPlus_iot.py
Traceback (most recent call last):
  File "bin/upsPlus_iot.py", line 18, in <module>
    ina_supply = INA219(0.00725,address=0x40)
  File "/home/teste/.local/lib/python3.7/site-packages/ina219.py", line 116, in __init__
    self._i2c = I2C.get_i2c_device(address=address, busnum=busnum)
  File "/home/teste/.local/lib/python3.7/site-packages/Adafruit_GPIO/I2C.py", line 63, in get_i2c_device
    busnum = get_default_bus()
  File "/home/teste/.local/lib/python3.7/site-packages/Adafruit_GPIO/I2C.py", line 55, in get_default_bus
    raise RuntimeError('Could not determine default I2C bus for platform.')
RuntimeError: Could not determine default I2C bus for platform.
teste@rpi4-20210718:~$ i2cdetect -y 1
-bash: i2cdetect: command not found
teste@rpi4-20210718:~$ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f
20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f
30: -- -- -- -- -- -- -- -- 38 39 3a 3b 3c 3d 3e 3f
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f
70: 70 71 72 73 74 75 76 77
```

hellresistor, 2021-07-24:
Well a little search and find this !!

```
teste@rpi4-20210718:~$ sudo i2cdetect -y 0
[sudo] password for teste:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f
20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f
30: -- -- -- -- -- -- -- -- 38 39 3a 3b 3c 3d 3e 3f
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f
50: 50 51 -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f
70: 70 71 72 73 74 75 76 77
teste@rpi4-20210718:~$ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f
20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f
30: -- -- -- -- -- -- -- -- 38 39 3a 3b 3c 3d 3e 3f
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f
70: 70 71 72 73 74 75 76 77
teste@rpi4-20210718:~$ sudo i2cdetect -y 2
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
teste@rpi4-20210718:~$ sudo i2cdetect -y 3
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- 17 -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: 40 -- -- -- -- 45 -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

seems the ic2-3 and not 1

hellresistor, 2021-07-25:
I can get information using [@frtz13](https://github.com/frtz13/UPSPlus_mqtt.git) script (setProtectionVoltage.py) Will get info and set with success.
```
/UPSPlus_mqtt# python3 setProtectionVoltage.py 3000
------------------------------------------------------------
Modify battery protection voltage in UPS Plus
------------------------------------------------------------
Current value:  3600 mV
Successfully set the protection voltage to: 3000 mV
```

Using your scripts... no .. :|  getting this errors
```
# python3 Full-featured-demo-code.py
Traceback (most recent call last):
  File "Full-featured-demo-code.py", line 20, in <module>
    ina_supply = INA219(0.00725, address=0x40)
  File "/usr/local/lib/python3.7/dist-packages/ina219.py", line 116, in __init__
    self._i2c = I2C.get_i2c_device(address=address, busnum=busnum)
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_GPIO/I2C.py", line 63, in get_i2c_device
    busnum = get_default_bus()
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_GPIO/I2C.py", line 55, in get_default_bus
    raise RuntimeError('Could not determine default I2C bus for platform.')
RuntimeError: Could not determine default I2C bus for platform.
```

hellresistor, 2021-07-28:
Solution: https://github.com/frtz13/UPSPlus_mqtt/issues/14#issuecomment-888541414
Maybe need do update on your code ?!

closed 2021-07-28 by hellresistor