geeekpi/upsplus #40: very good product, but there is a feature request.

layleon602, 2021-05-25:
I have been using this product for more than a month and it works well, there were two-times AC power outages and the system was not affected at all, but I recently tried the OTA function and I found that this function has to remove the battery in order to use it, which is a bit of a hassle and I hope it can be improved.

Another is that after the battery is full, it seems to be recharged every 5 minutes?

nickfox-taterli, 2021-05-27:
We are considering this issue, in addition to the interval flashing charging light is a sampling problem to avoid the occurrence of stopping charging.

layleon602, 2021-05-28:
I found that I can adjust the sample time to control this event, but if I want to adjust to a large sample time seems to lead to failure to charge?

nickfox-taterli, 2021-05-28:
This depends on the battery, and you can adjust this parameter dynamically.

closed 2021-06-10 by nickfox-taterli