geeekpi/upsplus #42: (In progress) 1st use. not get full charged.

hellresistor, 2021-05-30:
Hello, after the 'saga' about my first UPSplus device, me @deHarro have getting problems. 
I put it a side and start over 0. 

With a New RPI4 and the 2nd UPSpro (never used, and get power button OK).
 - Get Full Charged Batteries first thing
 - Assembled the UPSpro on RPI4 like official instrunctions (until now should be a FW5).
 - Insert a sdcard with clean raspberryOS (previously uncommented the line `dtparam=i2c_arm=on` on  `/boot/config.txt` file)
 - Insert Batteries ( UPS detect batteries but do not power ON, seems OK)
 - Insert Power Cable (UPS Start Charging, NOT start RPI, Until here, seems OK)
 - Get 3 fix blue LED and 1 Led Blink (suppose full charged...)
 - Keep it charging like 2hours (THIS IS WHEN I NOTICED, NOT when really happens, maybe less time than 2hours). All Leds Became Off during ~2 seconds, than start the charging process, Like Have Disconnect and Connect the cable.  and keep same loop between 3~5m.
 - Never Get Fully Charged, baterries suppose fully charged.
 
Waiting some feedback to can proceed with my test.

The Next Steps to do: (waiting):
 - Click UPSplus power Button
 - Finish Rasberry OS config and mannually add values `dtoverlay=i2c-rtc,ds1307` on `/boot/config.txt` file
 - Shutdown RPI
 - Remove PowerCable + Batteries
 - Insert Batteries + PowerCable
 - If Rpi Run Auto or need click ups power button (EXPECTED?!?) start the main test before booting, If get fully charged batteries LED indicator.
 - IF not Fix 4 LEDS. Boot Rpi again and DO the Fw Upgrade to fw7. And Re-Test again..

nickfox-taterli, 2021-05-30:
He is cyclic charging, every [sample cycle] will recheck whether the battery is full, and then re-charge, if you want to reduce the number of this, you can adjust [sample cycle].

Battery even if not used, he will also be slightly discharged, you can do an experiment, for example, you have a mobile power, you first give him a full charge, and then remove the charging cable, wait a while, and then plug in the charging cable, will also be re-charged.

Translated with www.DeepL.com/Translator (free version)

hellresistor, 2021-05-30:
Understand and Clear! Will Continue! Thank you!

hellresistor, 2021-05-30:
This Steps was reproduced. 
> The Next Steps to do: (waiting feedback):
> 
> * Click UPSplus power Button
> * Finish Rasberry OS config and mannually add values `dtoverlay=i2c-rtc,ds1307` on `/boot/config.txt` file
> * Shutdown RPI
> * Remove PowerCable + Batteries (EXPECTING?!)
> * Insert Batteries + PowerCable
> * If Rpi Run Auto or need click ups power button (EXPECTED?!?) start the main test before booting, If get fully charged batteries LED indicator.
> * IF not Fix 4 LEDS. Boot Rpi again and DO the Fw Upgrade to fw7. And Re-Test again..

Keep on same. Just some errors because FW. 
Next step Upgrade Firmware, launch time ^^

nickfox-taterli, 2021-05-31:
If you remove the battery, install the battery when you press and hold the button, will be forced to enter the method 1 of the OTA upgrade, which is to prevent some people in the process of upgrading the entire system damage, and re-upgrade the firmware through OTA.

I don't know if that's what you're describing.

closed 2021-06-10 by nickfox-taterli

hellresistor, 2021-06-10:
Well. i can update to close this topic.
I have one ups not good. because the other one (last used, working good. works like 1week without problem)
the first one keeps the drasticly stop charging and never charge batteries again (with fw8). i will contact the after sales.

deHarro, 2021-06-10:
@hellresistor 
Please keep us updated on how your second (good) UPS behaves in the future. Thanks!
Harry

hellresistor, 2021-06-14:
> @hellresistor
> Please keep us updated on how your second (good) UPS behaves in the future. Thanks!
> Harry

on this week will set that 2nd Ups on rpi 4GB with HomeAssistant and some smart-plugs devices connected and some NodeRed flows. and yes. a real test. :) i will keep it updated. (past week i have been very busy)