geeekpi/upsplus #47: Correct line for downgrading f/w?

ArjenR49, 2021-06-17:
There's a line in the OTA_firmware_upgrade-py script which can be used to downgrade the f/w to an older version:
r = requests.post(UPDATE_URL, data={"UID0":UID0, "UID1":UID1, "UID2":UID2, , "ver":7})

The extra comma before the "ver" has me wondering. Is it correct?
There seems to be an empty item first and only then the "ver":7

I'd rather not blindly try to find out, which might leave the UPS in the upgrade mode.

leandroalbero, 2021-06-17:
I believe you have to remove the extra comma, if you try to run it as is you get the following:
```
leandroalbero@Leandros-MacBook-Pro ~ % python3 test2.py
  File "/Users/leandroalbero/test2.py", line 31
    r = requests.post(UPDATE_URL, data={"UID0":"", "UID1":"", "UID2":"",, "ver":7})
                                  ^
SyntaxError: expression cannot contain assignment, perhaps you meant "=="?
```

ArjenR49, 2021-06-17:
Thanks!
The extra comma was flagged by Sublime, the editor ...

I had to look up 'lint some lines'. Hadn't seen that word in a figurative sense.
So Sublime linted the code ... (It runs on my Linux Mint laptop).

ArjenR49, 2021-06-18:
I had to do the f/w downgrade from v.8 to v.7 using method 2 for setting the upgrade mode, i.e. remove batteries & external power, press button & keep it pressed, insert batteries ...
After using method 1 the up-/downgrade script would exit with an error.

nickfox-taterli, 2021-06-19:
Only some versions are available for rollback, if the version that cannot be rolled back will return HTTP 404, and what is the specific error?

ArjenR49, 2021-06-19:
I didn't save the output and I am not going to try again. Sorry. But I repeated it several times, because i2c errors seem to happen some times and then not.

closed 2021-06-20 by nickfox-taterli