geeekpi/upsplus #49: Battery voltage never above 3.8V

olonsoft, 2021-06-23:
New customer here.
I notice that the battery never charges above 3.8V and the 4th led always blinks.
Is that normal?
Report from `upsPlus.py`
```
------------------------------------------------------------
------Current information of the detected Raspberry Pi------
------------------------------------------------------------
Raspberry Pi Supply Voltage: 4.980 V
Raspberry Pi Current Current Consumption: 514.382 mA
Raspberry Pi Current Power Consumption: 2479.394 mW
------------------------------------------------------------
-------------------Batteries information-------------------
------------------------------------------------------------
Voltage of Batteries: 3.820 V
Battery Current (Charging) Rate: 86.098 mA
Current Battery Power Supplement: 336.585 mW
Successfully set the protection voltage to: 3700 mV
------------------------------------------------------------
Currently charging via Micro USB Port.
```
Using a 5V 3A Raspberry certified charger.
I am using firmware v7. Downgraded from v8 because of instability problems with that version.

frtz13, 2021-06-23:
please run the Full-Featured-Demo-Code.py script (you may want to comment out the 3 bus.write_byte... lines in the last part of the script). there are two measurements of the battery voltage available. one by a INA219 sensor, another is available in registers 6 and 5 (I guess this is from the IP5328P charge controller). both show the same values? do you have a voltmeter at hand?

olonsoft, 2021-06-24:
Something must be wrong with firmware or hardware.
While the report shows 3.81V, I measure with a multimeter the batteries and are 4.2V.
I then unplug the power, remove and add the batteries again. Now the python script shows 4.196V
The "Full-featured-demo-code.py" shows:
```
Raspberry Pi power supply voltage: 4.980 V
Current current consumption of Raspberry Pi: 535.071 mA
Current power consumption of Raspberry Pi: 2661.060 mW
Batteries Voltage: 4.196 V
Battery current (charging), rate: 7.073 mA
Current battery power supplement: 87.805 mW
Current processor voltage: 3286 mV
Current Raspberry Pi report voltage: 5005 mV
Current battery port report voltage: 4183 mV
Current charging interface report voltage (Type C): 3 mV
Current charging interface report voltage (Micro USB): 5186 mV
Currently charging via Micro USB.
Current battery temperature (estimated): 49 degC
Full battery voltage: 4296 mV
Battery empty voltage: 3939 mV
Battery protection voltage: 3600 mV
Battery remaining capacity: 99 %
Sampling period: 2 Min
Current power state: normal
No shutdown countdown!
Automatically turn on when there is external power supply!
No restart countdown!
Accumulated running time: 53280 sec
Accumulated charged time: 53280 sec
This running time: 53280 sec
Version number: 7
Serial Number is: xxxxxxxx-xxxxxxxx-xxxxxxxx
```
Since first installation I 've never seen the 4rth charging (blue) led to have stopped blinking.

ArjenR49, 2021-06-24:
7 mA charging current is lower than what I have ever seen on my UPSplus ...

(mine has two batteries of some 3400 mAh each, Sanyo, locally bought)

frtz13, 2021-06-24:
@olonsoft This is strange, indeed. I regularly experience a situation where the UPS registers "freeze" (i.e. always return the same values) after some days. This happened with several firmware versions I tried so far (see issue #28 for ex.). When this happens, the INA219 sensors on my board continue to work as usual. However, in your case, it seems that the faulty voltage reading comes from one of the INA219 sensors.

yoyojacky, 2021-06-30:
> Something must be wrong with firmware or hardware.
> While the report shows 3.81V, I measure with a multimeter the batteries and are 4.2V.
> I then unplug the power, remove and add the batteries again. [...]

It must not be fully charged. Think about it, your Raspberry Pi is consuming power and your power supply is charging. Your charging speed will never keep up with your discharging speed. There will definitely be a situation of permanent dissatisfaction!
And the state of different batteries is different when charging. Unless when you have an external power supply, the external power supply takes over your Raspberry Pi, and then another way to charge the battery.

yoyojacky, 2021-06-30:
> […](#)

Your battery doesn't seem to be charged.

yoyojacky, 2021-06-30:
> New customer here.
> I notice that the battery never charges above 3.8V and the 4th led always blinks.
> Is that normal?
> Report from `upsPlus.py`
>
> ``` [...]

Can you replace this set of batteries and try another set of batteries?

yoyojacky, 2021-06-30:
> r hardware.
> While the report shows 3.81V, I measure with a multimeter the bat

That is real-time sampling information. You can sample several times and then take the average value. A discrete sampling data is not enough to indicate that the device firmware or equipment is faulty. In addition, the data information read by the multimeter and the information sampled by the INA219 will be slightly different. Yes, because the reference voltage is also different.

yoyojacky, 2021-06-30:
> @olonsoft This is strange, indeed. I regularly experience a situation where the UPS registers "freeze" (i.e. always return the same values) after some days. This happened with several firmware [...]

It is better to check whether there are multiple programs executing in the background or reading and writing data to the registers at the same time? Is the system busy?
Is the CPU load high?

frtz13, 2021-06-30:
only one single program reads the UPS registers, so no concurrent access. the RPi 4B is not busy at all, running without fan cooling with CPU at 52°C. I am currently running firmware v.7 and the freeze starts after about 2 days of operation. your IOT platform currently gets telemetry data from this "frozen" system. look at 'Feed!','id':1416891 for ex.

olonsoft, 2021-06-30:
> It is better to check whether there are multiple programs executing in the background or reading and writing data to the registers at the same time? Is the system busy?
> Is the CPU load high?

First of all, I have to tell you that I am using Raspberry PI 3 and not 4. 3 has less power consumption than 4. The power supply I am using is 3A (tested). I think that's enough to fully charge the batteries and power the RPi.
There are no other programs reading and writing to the registers at the same time, nor the system is busy.
For more than 2 weeks that my UPS is working, the 4rth blue (charging) led is always blinking.

ArjenR49, 2021-07-11:
It does that blinking all the time when operating with a load.
You may have noticed the blue lights go off every now and then. If you check the various variables when the blue lights are off, you should see that the batteries are being discharged with the load of the Pi at that moment. I think it is because the UPS collects information about the state of the batteries. While the lights are off, the batteries don't get charged and all current to the Pi comes from the batteries. This is why the batteries are being charged 'all the time' and the fourth blue led blinks.
When the UPS plus sits without a load (Pi off), the blue lights eventually stop blinking. I observed this, but naturally the Pi is really almost always running when the UPS/Pi combination is powered.

At least this is what I observed and how I explain my observations to myself. I don't think this has been clearly documented as such. I could be wrong on details or perhaps completely, so check for yourself.
The charging current falls below 100 mA with my batteries (3500mAh, quality cells, bought locally in Finland).

I'm not using the UPS just now. I have reverted it to v.7 and it works fine, AFAIK, but I am now setting up other Pi's in a folding@home and rosetta@home cluster. The Pi with the UPS plus is intended as a 'hub' and eventually also motioneye camera server. I'll set it up again later in its 'final' constellation, including the UPS and Pi camera with some IR lights ...