geeekpi/upsplus #50: Battery voltage reports 32V

ggtools, 2021-06-24:
I brought a new UPS and while everything is OK I have a strange battery voltage which always reports 32V. I didn't check this with a voltmeter but I'm positively sure that two 3.7 v battery cannot output 32V even by accident.

The output of Full-featured-demo-code.pyis:

```
Raspberry Pi power supply voltage: 5.004 V
Current current consumption of Raspberry Pi: 1198.654 mA
Current power consumption of Raspberry Pi: 5991.590 mW
Batteries Voltage: 32.000 V
Battery current (charging), rate: 2615.854 mA
Current battery power supplement: 83712.195 mW
Current processor voltage: 3337 mV
Current Raspberry Pi report voltage: 5032 mV
Current battery port report voltage: 3857 mV
Current charging interface report voltage (Type C): 6 mV
Current charging interface report voltage (Micro USB): 0 mV
Currently not charging.
Current battery temperature (estimated): 39 degC
Full battery voltage: 4297 mV
Battery empty voltage: 3700 mV
Battery protection voltage: 3700 mV
Battery remaining capacity: 18 %
Sampling period: 2 Min
Current power state: normal
No shutdown countdown!
Automatically turn on when there is external power supply!
No restart countdown!
Accumulated running time: 59847 sec
Accumulated charged time: 283677 sec
This running time: 3931 sec
Version number: 8
Serial Number is:private
```

yoyojacky, 2021-06-30:
please execute the python script again ? Is it remain this output?

ggtools, 2021-07-06:
Sorry I was away so the whole RPi/UPS was off. I let it run a couple of days being plugged and I still get the following output:

```Raspberry Pi power supply voltage: 5.008 V
Current current consumption of Raspberry Pi: 1176.451 mA
Current power consumption of Raspberry Pi: 5890.664 mW
Batteries Voltage: 32.000 V
Battery current (charging), rate: 4127.805 mA
Current battery power supplement: 134141.463 mW
Current processor voltage: 3312 mV
Current Raspberry Pi report voltage: 5001 mV
Current battery port report voltage: 4197 mV
Current charging interface report voltage (Type C): 8938 mV
Current charging interface report voltage (Micro USB): 3 mV
Currently charging through Type C.
Current battery temperature (estimated): 41 degC
Full battery voltage: 4303 mV
Battery empty voltage: 3700 mV
Battery protection voltage: 3700 mV
Battery remaining capacity: 90 %
Sampling period: 2 Min
Current power state: normal
No shutdown countdown!
Automatically turn on when there is external power supply!
No restart countdown!
Accumulated running time: 315702 sec
Accumulated charged time: 1273354 sec
This running time: 125138 sec
Version number: 8
```

closed 2024-04-08 by yoyojacky