geeekpi/upsplus #51: Change output voltage

Gieke85, 2021-06-29:
With both my portable monitor and SSD (sata-to-usb) connected the output voltage becomes 0.1v to low for the SSD to function. 

Is there a way to increase the pogopin output voltage?

frtz13, 2021-06-29:
I don't think you can do it by programming. as far as I know, the component responsible for the output (U1) is a TPS61088A boost converter. the voltage divider in the feedback path is made up by R3 and R5. values are 200kOhm 1% resp. 69.8kOhm 1% (see issue #26). the output voltage on my board also is about 4.9V at 1.3A charge. If ever this becomes a problem for my set-up I may go and adjust this voltage divider, but so far I did not.

ArjenR49, 2021-06-29:
Nickfox-taterli gave R3 as 220k, not 200k. 

I may try to raise the voltage a bit, too, IF I can get at R5 comfortably and add a small conventional wired resistor of a suitably large value (1200k if R3 is 220k) in parallel without desoldering R5 by accident.

yoyojacky, 2021-06-30:
> With both my portable monitor and SSD (sata-to-usb) connected the output voltage becomes 0.1v to low for the SSD to function.
> 
> Is there a way to increase the pogopin output voltage?

According to this structure, your idea cannot be realized.
[image]

closed 2021-06-30 by yoyojacky