geeekpi/upsplus #54: V8 Firmware overcharged Samsung 25R batteries

nikivanov, 2021-07-09:
I'm running V8 Firmware. I left the unit plugged into the charger overnight. When I came back it luckily didn't catch fire. However both batteries, genuine Samsung 25R, are COMPLETELY dead - multimeter can't detect any appreciable voltage between the terminals. I've unplugged the charger and replaced the batteries with another fully charged set. When I ran the Full-featured-demo-code.py script it gave me:

Full battery voltage: 4914 mV

It seems the unit kept overcharging them until some internal fuse blew. I tried doing the following:

fullVoltage = 4100
self.bus.write_byte_data(DEVICE_ADDR, 13, fullVoltage & 0xFF)
self.bus.write_byte_data(DEVICE_ADDR, 14, (fullVoltage >> 8)& 0xFF)

but it doesn't seem to be able to write to that register - Full voltage is still at 4.9V

Has anyone else had a similar issue? Is it even possible to set max charging voltage? I think it should be - batteries last longer if they aren't charged to 100%, so maybe setting max charged V to 4.0V might be useful.

hellresistor, 2021-07-09:
what miliamper batteries value 2500mAh ?

nikivanov, 2021-07-09:
Yep, 2500mah

hellresistor, 2021-07-10:
> Yep, 2500mah

Mine 2600mah. I am waiting arrive of 2000mah batteries to test...

nikivanov, 2021-07-10:
To test what? The unit should stop charging batteries once they reach a certain voltage. This used to work in earlier versions of the firmware, fwiw.

hellresistor, 2021-07-10:
batteries quality. other brand and other values.

nickfox-taterli, 2021-07-30:
in fact, the capacity of the battery is determined by the battery, modifying this parameter is to determine the charging behavior, when close to full charge voltage, it will actively reduce the charging rate.

nikivanov, 2021-08-04:
@nickfox-taterli Here's the output of the demo script:

~/upsplus $ python3 Full-featured-demo-code.py 
Raspberry Pi power supply voltage: 4.976 V
Current current consumption of Raspberry Pi: 478.553 mA
Current power consumption of Raspberry Pi: 2624.054 mW
Batteries Voltage: 0.000 V
Battery current (discharge), rate: -895.854 mA
Current battery power consumption: 0.000 mW
Current processor voltage: 3360 mV
Current Raspberry Pi report voltage: 5034 mV
Current battery port report voltage: 4619 mV
Current charging interface report voltage (Type C): 0 mV
Current charging interface report voltage (Micro USB): 0 mV
Currently not charging.
Current battery temperature (estimated): 43 degC
Full battery voltage: 4000 mV
Battery empty voltage: 3000 mV
Battery protection voltage: 3000 mV
Battery remaining capacity: 99 %
Sampling period: 2 Min
Current power state: normal
No shutdown countdown!
Does not automatically turn on when there is an external power supply!
No restart countdown!
Accumulated running time: 2777 sec
Accumulated charged time: 150678 sec
This running time: 414 sec
Version number: 9 

I left the UPS unit plugged in for too long (last time I did that it killed my batteries altogether). This time it just overcharged the batteries. Note the battery port voltage - it's 4.6V. These are genuine Samsung 25R batteries.

closed 2024-04-08 by yoyojacky