geeekpi/upsplus #55: i2cset shutdown doesn't work if on AC

b4dpxl, 2021-07-16:
I created a shutdown script to power off the UPS if the Pi is powered off:

`/lib/systemd/system-shutdown/ups-poweroff`:

```shell
#! /bin/bash

if [ "x$1" == "xpoweroff" ] || [ "x$1" == "xhalt" ] ; then
    # power off the UPS in 5 seconds
    /usr/sbin/i2cset -y 1 0x17 24 5
fi
```

This works fine if the UPS is on battery, but if the UPS is on AC power if doesn't work. This can also be seen by running `i2cset -y 1 0x17 24 60` and observing `watch -n 1 i2cget -y 1 0x17 24`; when on battery the value decreases, but not when on AC. I'm running the latest firmware.

Also, if you use the button to shut off the Pi, if "back-to-ac" is enabled, the Pi turns back on after a few seconds.

nickfox-taterli, 2021-07-30:
sw V8 after the repair of this problem, when you enter the shutdown command, he will first shut down, and then determine whether there is AC power, if so, will be turned on, if you want to completely shut down no longer on, you need to shut down the BACK-TO-AC.

sfatula, 2022-11-21:
So, is there no way to actually power off the UPS when on AC? I set register 25,26 to 0, and 24 to 10, and, while machine does not reboot, I notice power is still supplied to attached devices such as USB boot drive.

Is there no way to turn off the power supplied by the UPS entirely when AC still exists?

galtobellojr, 2022-11-22:
There was a discussion here regarding the USB 5V remaining active when unit
is "powered off", and another noted that batteries are completely drained
when powered off and disconnected from AC (in my experience, a week or two
does it).

I've not looked into it lately, but my conclusion has been that if the USB
port indeed remains live, then the battery exhaustion is explained (and,
truly, expected).

This disappointed my hope to have a Portable Pi on the shelf that could be
booted and remain running even while charging, much like the
portability/rechargeability of one's cellphone.  I've added tool-less
knurled screws to the battery cover, as the batteries shouldn't stay in the
unit when stored. Minor aggravation in my use case, not a total loss.

Just cleared my desk for the winter, and plan to see what happens when I
connect mains power to that drained unit. I have a fireproof bag handy...

I don't have the thread IDs handy, so hope what's here helps you.

   -gus

sfatula, 2022-11-23:
I'm not using the code here, just my own version of it heavily modified. When I tested with no power supply, the UPS does shut off, and goes dark. It just won't do that with a supply hooked up, say I want to shut it down while out of town. I can't really.

closed 2024-04-08 by yoyojacky