geeekpi/upsplus #56: Battery voltage bad data after a few hours.

peacho10, 2021-07-21:
Hi, after being in operation for a few hours, the battery voltage reading is wrong.

`**2021/07/21 14:05:02**
Raspberry Pi power supply voltage: 4.760 V
Current current consumption of Raspberry Pi: 2081.245 mA
Current power consumption of Raspberry Pi: 10247.267 mW
**Batteries Voltage: 4.220 V**
Battery current (charging), rate: 168.049 mA
Current battery power supplement: 712.195 mW
Current processor voltage: 3309 mV
Current Raspberry Pi report voltage: 5116 mV
Current battery port report voltage: 4187 mV
Current charging interface report voltage (Type C): 5417 mV
Current charging interface report voltage (Micro USB): 38 mV
Currently charging through Type C.
Current battery temperature (estimated): 48 degC
Full battery voltage: 4270 mV
Battery empty voltage: 3789 mV
Battery protection voltage: 3600 mV
Battery remaining capacity: 97 %
Sampling period: 2 Min
Current power state: normal
No shutdown countdown!
Automatically turn on when there is external power supply!
No restart countdown!
Accumulated running time: 7978 sec
Accumulated charged time: 8002 sec
This running time: 7972 sec
Version number: 9
Serial Number is: xxxxxxx-xxxxxxx-xxxxxxx
`
1 minute later:

`**2021/07/21 14:06:01**
Raspberry Pi power supply voltage: 4.972 V
Current current consumption of Raspberry Pi: 2510.177 mA
Current power consumption of Raspberry Pi: 13917.578 mW
**Batteries Voltage: 3.764 V**
Battery current (charging), rate: 128.049 mA
Current battery power supplement: 448.780 mW
Current processor voltage: 3309 mV
Current Raspberry Pi report voltage: 5094 mV
Current battery port report voltage: 4214 mV
Current charging interface report voltage (Type C): 5391 mV
Current charging interface report voltage (Micro USB): 64 mV
Currently charging through Type C.
Current battery temperature (estimated): 48 degC
Full battery voltage: 4270 mV
Battery empty voltage: 3789 mV
Battery protection voltage: 3600 mV
Battery remaining capacity: 97 %
Sampling period: 2 Min
Current power state: normal
No shutdown countdown!
Automatically turn on when there is external power supply!
No restart countdown!
Accumulated running time: 8034 sec
Accumulated charged time: 8058 sec
This running time: 8028 sec
Version number: 9
`
It also happens with firmware 7 and 8.

The only way to get a good measurement again is to turn off the UPS, remove the batteries, put the batteries and turn the UPS back on.

ArjenR49, 2021-07-21:
Where did you get f/w version number 9?

On geeekpi's GitHub there's only v. 8 ...

I went back from 8 to 7 weeks ago, because I had big problems with v. 8.
Couldn't write to registers meant to be written to.
So I've been waiting for an update all this time or at least an
acknowledgement of the problems.

My only problem now with version 7 is that if you shut down the pi from its
OS, you cannot start it again unless you take out the batteries. The button
is dead.
But if the power is disconnected and the pi stops after a while, it will
restart when power is restored. That much works fine now. I use frtz13's
software from GitHub.

I plan to reset the ups to factory defaults one day soon and see if that
solves the button problem.
Reset is mentioned in the table of the documentation ep-0136. Write a 1 to
address 0x1B ... Don't remember if I ever tried that before.

Suggestion for development:
Since there is a RTC on the ups board, one could expect there to be a
possibility to program an automatic boot. RTC's seem to have the
possibility to do a particular something at a certain time.  I have an
Olmatic UPS on a pi3b+  which uses that functionality of the RTC if so
desired.

Arjen
(On The Road)

hellresistor, 2021-07-21:
@ArjenR49 I got on my last firmware upgrade the V9

ArjenR49, 2021-07-21:
@hellresistor You will have noticed that these GitHub don't know a thing about this version 9. I have been checking these pages for weeks and hoping for a newer and hopefully better version.
So, what would you be able to say about version 9, especially compared to v.8, which had serious issues, in my experience, at least. I had to go back to v.7, once I found out it is possible at all.

closed 2024-04-08 by yoyojacky