geeekpi/upsplus #60: [Firmware v.9] Protection Voltage Reset (and other parameters)

ClaxiVa, 2021-07-27:
So basically for some reason the Protection voltage just resets itslef to the default 3700mV once in a minute, exactly at 01 second of every minute.

Also what's the point in setting Protection voltage same as Empty voltage?

And while in case with Protection voltage I'm able to rewrite it for at least a minute, I can do nothing with Empty voltage, it accepts arguments through i2c but the result is the same – 3700mV for Empty voltage. That's why my RasPi4 turns off when I got 2 LEDs on saying my UPS is only half-way epmty, lol

And the same goes to the Sampling period btw – unable to change it.

https://user-images.githubusercontent.com/59376597/127082337-b516546d-4cc8-4886-9ecf-fc9690272e5a.mp4

ClaxiVa, 2021-07-28:
Ok so I tried to roll back to v.7 (many pointed out is was better), now it doesn't rewrite UPS's protection voltage, but the empty voltage stays at 3700mV. Is there a way to change it at all?

ClaxiVa, 2021-07-28:
I tried to roll back to v7 and it still behaves strangely (once in couple of minutes it rewrites Protection voltage to 3700 but then rewrites it back to the value I typed in by itself), but at least Protections voltage seems fine. Yet Empty voltage still stays at 3700 and it pisses me off: judging by LEDs the UPS knows the batteries are only half-way empty, but anyways the system turns off at 3700mV

nickfox-taterli, 2021-07-30:
There may be default scripts in the crontab that are modifying this parameter.

ClaxiVa, 2021-07-30:
I've checked that, there were some in /bin directory (upsPlus and upsPlus_iot), but even after removing those from crontab and rebooting the system Protection voltage rools back to 3700 every minute

bchwtz, 2021-08-07:
I had the same issue due to the cronjobs being called both from user `pi `and `root`. Disabling (just commenting them out) resolved this problem. Here is what my `crontab` looks like now.

```
pi@raspberrypi:~ $ crontab -l
#* * * * * /usr/bin/python3 /home/pi/bin/upsPlus.py > /tmp/upsPlus.log
#* * * * * /usr/bin/python3 /home/pi/bin/upsPlus_iot.py > /tmp/upsPlus_iot.log
pi@raspberrypi:~ $ sudo crontab -l
# ROOT Crontab
#* * * * * /usr/bin/python3 /root/bin/upsPlus.py > /tmp/upsPlus.log
#* * * * * /usr/bin/python3 /root/bin/upsPlus_iot.py > /tmp/upsPlus_iot.log
```

ClaxiVa, 2021-08-09:
> I had the same issue due to the cronjobs being called both from user `pi `and `root`. Disabling (just commenting them out) resolved this problem. Here is what my `crontab` looks like now.
>
> ``` [...]

Thank you, I didn't know crontab has two "instances". I'll see later if it works for me or not

closed 2024-04-08 by yoyojacky