geeekpi/upsplus #61: The last power LED is still flashing after the power is turned on all night.

HouJianWei610, 2021-07-28:
The last power LED is still flashing after the power is turned on all night. Does that mean the battery is still not fully charged?

ArjenR49, 2021-07-28:
You must have noticed that every once in a while all blue LED's go off for some time. If you would read the battery and Pi current value at such time, you may see that at that time the batteries are not charged, but discharged, and obviously feed the Pi.
I think the firmware collects data about the batteries during that period.
As the batteries get discharged somewhat during that period, they keep being charged at other times, so the 4th LED never stops blinking.
I have seen the fourth LED stopped blinking in some circumstance. I think it was when I shut down the Pi from the GUI with the charger left connected.
This is my understanding of this based on my own observations and some comment somewhere by GeeekPi.

HouJianWei0610, 2021-07-29:
thanks for youru replay,If this is normal, then I can rest assured

nickfox-taterli, 2021-07-30:
This is normal, but you can adjust the sampling period of the battery to control the charging method.

HouJianWei0610, 2021-08-02:
I re-tested in the shutdown state, still the same

ArjenR49, 2021-08-02:
Hi, I happened to check just yesterday shutting down the pi from the GUI.

Next I cut AC power to the ups. The red led on the pi was still on and the
blue lights were now all 4 lit, none blinking.

Then I pressed the ups button. This cut battery power to the pi, and the
red led went off.

After some short time - I didn't take the time - also the blue leds went
off by themselves, so the pi and the ups were then both dark.

Reconnecting the AC power started the pi up again.

Arjen
(On The Road)

HouJianWei0610, 2021-08-06:
Thank you for your reply，I mean when the UPS is not supplying power to the Raspberry Pi (turn off the UPS) and keep the AC power on all night, the last LED still flashes，Does this mean that the battery is still not fully charged?According to my understanding, charging should stop and the LED will stop blinking, because there is no power output and only input

ArjenR49, 2021-08-06:
If I shut down the Pi using e.g. the OS GUI, the UPS does not cut the power even if I wait some minutes. I think it is determined by a register setting.
If I then press the UPS button, the UPS cuts the power to the Pi for a few seconds and then supplies power again and therefore the Pi starts up. AC is connected all the time.

If you disconnect AC after shutting down the Pi, the button has a different effect. It will cut the power to the Pi and not start it up by reconnecting the power (from the batteries). In this case, the blue leds stop blinking and will soon go off, so the UPS and the Pi are totally dark.
Reconnecting AC power will start up the Pi again.

In my case the registers that determine the behaviour of the UPS towards the Pi are as follows when the Pi is running:

Current values of the UPS power control registers (excerpt from my own reporting script output):
0x18=0 / 0x19=1 / 0x1A=0
Explanation:
0x18 - Power off timer (no restart)   - not set.
0x19 - Automatic restart upon return of external power: yes
0x1A - Power off timer (with restart) - not set.

Manipulating these registers will change what happens when the Pi is shut down by the UPS - actually it is of course shut down by the control script. I run frtz13's fanShutDownUps.py and haven't added any code to the script to manipulate the registers 0x18-0x1A at the time of shut down (because the script determines that the battery voltage is too low).

When shutting down the Pi the script executes the following:
def Shutdown(mqttclient):  
    if controlFan:
        oFan.fanOFF()
    MQTT_Terminate(mqttclient)
    # initialize shutdown sequence.
    try:
        # enable switch on when back on AC 
        i2c_bus.write_byte_data(DEVICE_ADDR, 0x19, 1)
    except Exception as exc:
        syslog.syslog("[Shutdown] Error writing UPS register (back to AC power up): " + str(exc))
    try:
        i2c_bus.write_byte_data(DEVICE_ADDR, 0x18, SHUTDOWN_TIMEOUT)
    except Exception as exc:
        syslog.syslog("[Shutdown] Error writing UPS register (shutdown timeout): " + str(exc))
    time.sleep(1)
#    os.system("sudo shutdown -h 1")
    os.system("sudo sync && sudo halt")
    while True:
        time.sleep(100)    

It sets 0x19 to 1 and 0x18 to SHUTDOWN_TIMEOUT, which is defined as follows:
# ups will shut down RPi the power supply with this delay after OS shut down command was given
SHUTDOWN_TIMEOUT_s = 60

Before I found frtz13's excellent script, I did all sorts of tests manipulating these registers 0x18-0x1A in my own control script, and got interesting results, which then changed when the f/w changed and later I forgot about my test results. I have not used my own scripts any longer, except for reporting all kinds of UPS values in the terminal. I am happy with frtz13's script, even though I cannot understand it fully, because of the OOP type of coding (I think that's what it is, instances and so on, almost esoteric, haha).

E.g. I made a control script so that it would shut down the Pi when a certain amount of time had passed after AC failure, say 10 minutes. I also had a control script for making the UPS reboot the Pi (probably through setting register 0x1A). At one time those worked with an old f/w. Not sure about now.

ArjenR49, 2021-08-06:
Sorry about the formatting. I forgot how to prevent this system thinking it knows better than me.

HouJianWei0610, 2021-08-09:
Thanks i got it，but a new problem appear,Sorry to trouble you，When I run Full-featured-demo-code.py, it prompts me no moudle smbus2, I tried to install it, but failed

closed 2024-04-08 by yoyojacky