geeekpi/upsplus #67: Stuck at firmware, can't load in normal mode

yagee, 2021-08-02:
Hello!
After reboot i2cdetect shows these digits, as I can understand - that means UPSPlus in firmware update mode.

pi@pihome:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- 18 -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: 40 -- -- -- -- 45 -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

If I try to run OTA_firmware_update.py I see this errors:

pi@pihome:~/upsplus $ python3 OTA_firmware_upgrade.py
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 159, in _new_conn
    (self._dns_host, self.port), self.timeout, **extra_kw)
  File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 80, in create_connection
    raise err
  File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 70, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 600, in urlopen
    chunked=chunked)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 343, in _make_request
    self._validate_conn(conn)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 841, in _validate_conn
    conn.connect()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 301, in connect
    conn = self._new_conn()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 168, in _new_conn
    self, "Failed to establish a new connection: %s" % e)
urllib3.exceptions.NewConnectionError: <urllib3.connection.VerifiedHTTPSConnection object at 0x75e977b0>: Failed to establish a new connection: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 638, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 398, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='api.52pi.com', port=443): Max retries exceeded with url: /update (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x75e977b0>: Failed to establish a new connection: [Errno 111] Connection refused'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "OTA_firmware_upgrade.py", line 29, in <module>
    r = requests.post(UPDATE_URL, data={"UID0": UID0, "UID1": UID1, "UID2": UID2})
  File "/usr/lib/python3/dist-packages/requests/api.py", line 116, in post
    return request('post', url, data=data, json=json, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 516, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='api.52pi.com', port=443): Max retries exceeded with url: /update (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x75e977b0>: Failed to establish a new connection: [Errno 111] Connection refused'))

Any suggestions?

Thank you!

yagee, 2021-08-03:
So, I've solved this problem by myself.

If you have the same problem, here steps to fix it:

1. Manually download `firmware.bin`
2. Create the new `firmwareUpdate.py` with this code:
```
with open("firmware.bin", "rb") as f:
    while True:
        data = f.read(16)
        for i in range(len(list(data))):
            bus.write_byte_data(0x18, i + 1, data[i])
        bus.write_byte_data(0x18, 50, 250)
        time.sleep(0.1)
        print('.', end='', flush=True)

        if len(list(data)) == 0:
            bus.write_byte_data(0X18, 50, 0)
            print('.', flush=True)
            print('The firmware upgrade is complete, please disconnect all power/batteries and reinstall to use '
                    'the new firmware.')
            os.system("sudo halt")
            while True:
                time.sleep(10)
```
3. Run `python3 firmwareUpdate.py`
4. Run `i2cdetect -y 1` 
[image]
If you see 17 - that's means yours UPSPlus is in normal mode.

closed 2021-08-03 by yagee

mmaga, 2021-12-11:
I've tried to using local firmware file upgrade. So, I got no errors, but I'm still stuck in the OTA Firmware Upgrade Mode (0x18) :-/

Lonewalker0007, 2022-01-17:
> So, I've solved this problem by myself.
>
> If you have the same problem, here steps to fix it:
>
> 1. Manually download `firmware.bin`
> 2. Create the new `firmwareUpdate.py` with this code:
>
> ```
> with [...]

Where do you download the firmware.bin?

Becaise i have the same problem.

yagee, 2022-01-17:
> Where do you download the firmware.bin?
> 
> Becaise i have the same problem.

There is no direct link to firmware, but I found a workaround.
If you want to download by yourself — it's a long story.
Feel free to contact with me via telegram (same username as here)

There is the firmware I downloaded today:
[firmware.zip](https://github.com/geeekpi/upsplus/files/7880933/firmware.zip)

Lonewalker0007, 2022-01-17:
> > Where do you download the firmware.bin?
> > Becaise i have the same problem.
> 
> @Lonewalker0007
> 
> There is no direct link to firmware, but I found a workaround. If you want to download by yourself — it's a long story. Feel free to contact with me via telegram (same username as here)
> 
> There is the firmware I downloaded today: [firmware.zip](https://github.com/geeekpi/upsplus/files/7880933/firmware.zip)

Now i am out of 0x18 and the ups is in 0x17. So its out of update status.  (Did not use the firmware) i ran manually the ota script in the upsplus folder.
But now i have the next problem, my pi won't boot when i power it via the ups. The green light on the pi blinks 2 times. Indicating it can't read the sd card. Witch is strenge because when i power the pi via its on usbC it just boots

yagee, 2022-01-17:
I think it's over my knowledge about UPSPlus, I've just shared solution to finish firmware update process.

yagee, 2022-01-17:
> Now i am out of 0x18 and the ups is in 0x17. So its out of update status. (Did not use the firmware) i ran manually the ota script in the upsplus folder. But now i have the next problem, my pi won't [...]

I think it's over my knowledge about UPSPlus, I've just shared solution to finish firmware update process.