geeekpi/upsplus #68: [Firmware v.9] Set sensible default values in Firmware to avoid potentially dangerous side effects

bchwtz, 2021-08-07:
I am currently working with the `UpsPlus v5.0 PCB 01b Revision` with `Firmware v.9` flashed to it and it seems that the firmware does not use appropriate default values. The values discussed in this repo (e.g. `PROTECT_VOLT = 3700`) are only properly configured using various python files. I would expect something like the following as default in the firmware **binary** so that usage is safe even though the recommended software is not installed and the UPS can be used safely out of the box with e.g. a read-only Linux.

**Device: 0x17**
- Register 0x11 - 0x12 : 3500 mV
- Register 0x15 - 0x16 : 2 Minutes

**The behavior of the firmware in the actual state is unclear and not (fully) documented so I am not sure what potential effects of the default parameters are.** However, I expect that with the default protection voltage the UPS never shuts down and drains the batteries to a potentially dangerous low level?

Can you please offer a firmware with updated defaults? I am happy to provide a PR, given that the source of the firmware is made available.

ArjenR49, 2021-08-08:
Frtz13's and my tests show that the UPS will cut off the load at the 
empty voltage value even without any control script running. The empty 
voltage is learned (adjusted while the batteries are discharged) from 
the protection voltage set by the user while the batteries are 
discharged (no AC). If there is no control script running, the UPS will 
just cut power to the load as soon as the empty voltage is reached.

The control script with its 200 mV margin safeguards the OS from 
sustaining damage by shutting down the Pi in good order and well ahead 
of the empty voltage being reached. Depending on settings and 
circumstances a sudden increase in load current might cause the actual 
battery voltage to drop below the empty voltage, which would trigger the 
UPS. It's not what you want to happen, so a sensible control script 
imposes a 200 mV safety margin ...

I agree that the UPS is not fully documented ...

bchwtz, 2021-08-08:
Thanks for providing those insights. The function of the control script is clear but irrelevant for the issue. The protection voltage has default `65535`. The defaults for full and empty voltage are also `65535`.

When the control script is never executed, the protection voltage stays at `65535`, which is unsafe. Properly configured default values would allow safe usage out of the box. Can you adjust the default values in the next FW release @nickfox-taterli ?

closed 2024-04-08 by yoyojacky