geeekpi/upsplus #69: not charging more than 65%

hellresistor, 2021-08-07:
```
# python3 Full-featured-demo-code.py
Raspberry Pi power supply voltage: 4.908 V
Current current consumption of Raspberry Pi: 1143.482 mA
Current power consumption of Raspberry Pi: 5618.167 mW
Batteries Voltage: 4.200 V
Battery current (charging), rate: 21.951 mA
Current battery power supplement: 97.561 mW
Current processor voltage: 3293 mV
Current Raspberry Pi report voltage: 4947 mV
Current battery port report voltage: 4226 mV
Current charging interface report voltage (Type C): 5056 mV
Current charging interface report voltage (Micro USB): 16 mV
Currently charging through Type C.
Current battery temperature (estimated): 45 degC
Full battery voltage: 4616 mV
Battery empty voltage: 3700 mV
Battery protection voltage: 3700 mV
Battery remaining capacity: 65 %
Sampling period: 3 Min
Current power state: normal
No shutdown countdown!
Does not automatically turn on when there is an external power supply!
No restart countdown!
Accumulated running time: 166276 sec
Accumulated charged time: 166278 sec
This running time: 162639 sec
Version number: 9
```

more than 18hours cable connected  and `Battery remaining capacity: 65 %` not  charge more of this 65%,.
Some reason ? :|  Just charge only 3% since yesterday.

Batteries: 2x  3.7Volt 2600mAh 5C

ArjenR49, 2021-08-08:
How come the full battery voltage is so high?

The charging current of 22 mA looks quite normal, so something is 
protecting your batteries despite the  far too high full battery voltage.

Op 07-08-2021 om 14:46 schreef hellresistor:
> |Full battery voltage: 4616 mV Battery empty voltage: 3700 mV Battery 
> protection voltage: 3700 mV Battery remaining capacity: 65 % |

peacho10, 2021-08-09:
Hi.

This battery remaining capacity its bad.  

It can be seen that the voltage of the batteries is 4.2v. They are fully charged.
`Batteries Voltage: 4.200 V`

But according to the code, the batteries are considered fully charged at 4.6V:
`Full battery voltage: 4616 mV`

The 'Full battery voltage' should be: 4200mV not 4616 mV

hellresistor, 2021-08-10:
> How come the full battery voltage is so high? The charging current of 22 mA looks quite normal, so something is protecting your batteries despite the  far too high full battery voltage. Op [...]

i dont know :|

hellresistor, 2021-08-10:
> Hi.
>
> This battery remaining capacity its bad.
>
> It can be seen that the voltage of the batteries is 4.2v. They are fully charged.
> `Batteries Voltage: 4.200 V`
>
> But according to the code, the [...]

what is the correct address register to change the value to 4200v ??
`i2cget -y 1 0x17 25`

ArjenR49, 2021-08-10:
Some days ago I tried to manipulate the battery full voltage, and even 
though the write operation seemed succesful, the value stored didn't change.

The only way to set full and empty voltages seems to be 'learning'.

First do a factory defaults reset.

I'd recommend starting with partly charged batteries. Set the protection 
voltage as you like, 3700 mV or 3200 mV ... Do not run any UPS control 
script. Use an OS which you don't mind damaging, if you're unlucky. 
Prepare a script to report on the essential register content (I should 
upload my latest version to GitHub ...) . You can run this script now 
and then in a terminal during the learning process to check what is 
going on. Let the UPS charge the batteries. The full voltage will be 
pushed up little by little staying ahead of the actual voltage by some 
10's of mV's.

When you're satisfied the batteries are fully charged, disconnect the 
charger. Now the Pi will drain the batteries and the UPS will push the 
empty voltage down little by little staying below the actual voltage by 
some 10's of mV's. When the UPS f/w decides the batteries are empty 
(protection voltage/empty voltage), it will suddenly cut power to the 
Pi. Mostly the OS doesn't suffer damage from the sudden power cut.

You can repeat that charging and discharging process (without the 
factory reset, naturally) a few times and see how the full and empty 
voltage values settle to their proper levels when things are left to the 
UPS f/w alone.

Later you can of course run your control script and set the protection 
voltage to whatever you think suitable.

To avoid uncontrolled power cuts, control scripts need to implement a 
margin of 200 mV to stay away from the empty voltage even during sudden 
load upswings in case the protection voltage is set equal to the empty 
voltage.

Seconds ago I uploaded the latest version of my reporting script 
UPS_report_for_UPSPlus_mqtt.py to my GitHub (I apologize, it is super 
messy; I am not into GitHub; e.g. I applied changes made by Frtz13 to my 
own versions of his scripts using Kompare on my laptop ;-)

It's easy python, if you wonder what's in it ;-) I managed to make it 
use comma's as decimal separators. You learn something every day week ;-)

Good luck!

ArjenR49, 2021-08-10:
The battery full voltage occupies two bytes 0x0D-0x0E.
I wrote the script below based on another script, but the full voltage "doesn't take":

#!/usr/bin/python3
# -*- coding: utf-8 -*-

# adapted from scripts provided at GitHub: Geeekpi/upsplus by nickfox-taterli
# modified from script by @frtz13: ar - 08-06-2021, 03-08-2021, 04-08-2021

# This script behaves as if it writes the new value, but in reality nothing changes ...

import sys
import smbus2

# Define I2C bus
DEVICE_BUS = 1

# Define device i2c slave address.
DEVICE_ADDR = 0x17

print("-"*60)
print("Modify batteries full voltage for UPS Plus")
print("-"*60)

FV_Mini_mV = 4000
FV_Maxi_mV = 4325

# Raspberry Pi communicates with MCU via i2c protocol.
bus = smbus2.SMBus(DEVICE_BUS)
currentFullVoltage_mV = bus.read_byte_data(DEVICE_ADDR, 0x0E) << 0o10 | bus.read_byte_data(DEVICE_ADDR, 0x0D)
print("Current value:  %d mV" % currentFullVoltage_mV)

if len(sys.argv) > 1:
    try:
        givenFV_mV = int(sys.argv[1])
        if givenFV_mV >= FV_Mini_mV and givenFV_mV <= FV_Maxi_mV :
            bus.write_byte_data(DEVICE_ADDR, 0x0D, givenFV_mV & 0xFF)
            bus.write_byte_data(DEVICE_ADDR, 0x0E, (givenFV_mV >> 0o10) & 0xFF)
            print("Successfully set the batteries full voltage to: %d mV" % givenFV_mV)
        else:
            errMsg = "Valid batteries full voltage values range from {:.0f} to {:.0f} mV".format(FV_Mini_mV, FV_Maxi_mV)
            print(errMsg)
    except Exception as exc:
        print("Incorrect parameter: {}. ({})".format(sys.argv[1], str(exc)))
else:
    print("Usage: {} <batteries_full_voltage_in_mV>".format(sys.argv[0]))
    print("<batteries_full_voltage_in_mV> between {:.0f} and {:.0f}".format(FV_Mini_mV,FV_Maxi_mV))

closed 2024-04-08 by yoyojacky