geeekpi/upsplus #70: charge voltage limit user defineable

chickenherb, 2021-08-08:
I am using UPS-Plus fw V9 together with 2 Panasonic NCR18650B LiIon cells.

I am concerned about the voltage level of cells in the UPS-Plus.

In the datasheet the maximum charge voltage level for the cell is stated as 4,20 volt.

For enhanced lifetime for the LiIon cell it shoult be possible to limit the

charge voltage level including percentage, the protection limit and the empty limit by the user.

Even in the description for the UPS-Plus there is stated "Support 4.2V 4.35V 4.4V 4.5V lithium battery"

How do you select this ?

I did already write the battery full value to 4100 mV but the value is increasing already to 4285 mV 
operation during operation time (52945 seconds), which is out of specifications!

For an enhanced lifetime for the cells I would like to set the limits by myself, for example
the charge voltage to 4100 mV, which enhances lifetime and looses only very little amount of capacity.

You can keep the voltage level of the cell to the charge limit by switching temporary to the battery 
supply if the upper voltage limit is reached.

Here is an printout from my Full-featured-demo-code

Raspberry Pi power supply voltage: 5.104 V
Current current consumption of Raspberry Pi: 1657.864 mA
Current power consumption of Raspberry Pi: 8528.175 mW
Batteries Voltage: 4.192 V
Battery current (charging), rate: 166.098 mA
Current battery power supplement: 487.805 mW
Current processor voltage: 3308 mV
Current Raspberry Pi report voltage: 5121 mV
Current battery port report voltage: 4192 mV
Current charging interface report voltage (Type C): 8969 mV
Current charging interface report voltage (Micro USB): 51 mV
Currently charging through Type C.
Current battery temperature (estimated): 47 degC
Full battery voltage: 4285 mV
Battery empty voltage: 3200 mV
Battery protection voltage: 3200 mV
Battery remaining capacity: 95 %
Sampling period: 2 Min
Current power state: normal
Shutdown countdown: 95 sec
Automatically turn on when there is external power supply!
Restart countdown: 84 sec
Accumulated running time: 52845 sec
Accumulated charged time: 52845 sec
This running time: 52845 sec
Version number: 9 
Serial Number is:XXXXXXXX-XXXXXXXX-XXXXXXXX

Thanks in advance

chickenherb

ArjenR49, 2021-08-08:
I wouldn't set both countdown values at the same time and I also think 
that they only need to be set at the time the control script decides to 
shut down the Pi. By setting one of those registers the script is able 
to determine what the UPS is to do next after the control script has 
told the Pi to shut down and stopped executing.

The UPS will either count down to 0 and cut the power to the Pi, or

it will count down and cut the power to the Pi for a few seconds only, 
so the Pi will start up again.

That said, f/w versions have come and gone and things may have changed. 
I started learning about this UPS from f/w v.3 ...

Op 08-08-2021 om 11:30 schreef chickenherb:
> Shutdown countdown: 95 sec
> ...
> Restart countdown: 84 sec

chickenherb, 2021-08-08:
Hi ArjenR49,
thank you for your reply.
I am not shure about the purpose of the timers.
My idea is that the Shutdown Countdown Register is active after a shutdown event of the UPS ?
And the Restart Countdown register is active after all variables are back to normal after a shutdown condition and gives a 
delay to startup the raspberry again.
Thats why I want to give the Restart Register a delay for giving the battery a little bit of charge. But I did not
touch the registers until now.

 pi@raspberrypi:~/upsplus $ python3 Set_User_Voltages_Timers.py 
Full battery voltage: 4100 mV
Battery empty voltage: 3000 mV
Battery protection voltage: 3200 mV
Battery remaining capacity: 100 %
Successfully set the protection voltage as: 3200 mV
Successfully set the empty voltage as: 3000 mV
Successfully set the full battery voltage as: 4100 mV
Successfully set the state of charge as: 100 %
Successfully set the full shutdown timer to: 10 s
Successfully set the full restart timer to: 120 s
Full battery voltage: 4100 mV
Battery empty voltage: 3000 mV
Battery protection voltage: 3200 mV
Battery remaining capacity: 100 %
Shutdown timer: 10 s
Restart timer: 120 s
pi@raspberrypi:~/upsplus $ python3 Full-featured-demo-code.py 
Raspberry Pi power supply voltage: 5.136 V
Current current consumption of Raspberry Pi: 1377.965 mA
Current power consumption of Raspberry Pi: 7064.760 mW
Batteries Voltage: 4.192 V
Battery current (charging), rate: 106.098 mA
Current battery power supplement: 302.439 mW
Current processor voltage: 53744 mV
Current Raspberry Pi report voltage: 5093 mV
Current battery port report voltage: 4160 mV
Current charging interface report voltage (Type C): 9058 mV
Current charging interface report voltage (Micro USB): 51 mV
Currently charging through Type C.
Current battery temperature (estimated): 45 degC
Full battery voltage: 4100 mV
Battery empty voltage: 3000 mV
Battery protection voltage: 3200 mV
Battery remaining capacity: 100 %
Sampling period: 2 Min
Current power state: normal
Shutdown countdown: 10 sec
Automatically turn on when there is external power supply!
Restart countdown: 120 sec
Accumulated running time: 80739 sec
Accumulated charged time: 80739 sec
This running time: 80739 sec
Version number: 9 
Serial Number is:XXXXXXXX-XXXXXXXX-XXXXXXXX

Thanks in advance for your suggestions!

Chickenherb

ArjenR49, 2022-06-08:
On f/w v.10 if countdown register 0x1A is set to 60, the UPS will wait for 60 seconds before cutting power to the Pi. In those 60 seconds the Pi can perform an orderly shut down of the OS. After the shutdown (aka halt) the Pi just sits waiting for the power to be removed.
After the countdown timer (0x1A) has counted all the way to zero, power is removed from the Pi by the UPS.
Subsequently, in v.10, the UPS will **wait for about 10 minutes** before restoring power to the Pi, which will then start up its OS.
I haven't found a way to change this startup delay of 10 minutes.

This way the user can make the Pi do a power cycle even when there's no-one  physically present to push the button on the UPS board.

If instead of 0x1A, you set 0x18 to 60, the UPS will just cut power to the Pi after 60 seconds and you'd have to be present to push the button to restart the Pi.

I wrote two scripts, PowerCycle.py and PowerDown.py for the purposes described above. I intend to post them on my Github page.

closed 2024-04-08 by yoyojacky