geeekpi/upsplus #71: Shuts down when AC pulled and cannot turn back on

cat007mine, 2021-08-09:
Hi all.

Batteries are fully charged (although upsplus.py shows them at 98%).
When I pull the USB-C charge cable from the UPS, after about 1 minute the UPS shuts down and I cannot get it back on again with the button on the side.
The only way to get the UPS (and then the Pi) to come back on is to take the batteries out, and put them back in again.

cat007mine, 2021-08-09:
Also if I shut the pi down with     sudo shutdown -P now

I am unable to turn it back on with the button on the UPS. How do I turn the pi back on once I've shut it down?

kostaskas, 2021-11-02:
I also have the same problem, did you find a fix? I tried to reset to Factory Defaults, reupgraded firmware to v.10 but no luck, battery charges ok, shows 98%.

onxblog, 2021-11-26:
You need disable in (Full-featured-demo-code.py):
# Cancel automatic shutdown
bus.write_byte_data(DEVICE_ADDR, 24,0)
# and restart if you want
bus.write_byte_data(DEVICE_ADDR, 26, 0)

closed 2024-09-18 by yoyojacky