geeekpi/upsplus #72: raspberry pi 4/400 does not boot when powered from EP-0136

1-prop-head, 2021-08-09:


ArjenR49, 2021-08-09:
a wild guess: are all pogo pins of the UPS making proper contact with the Pi board?

1-prop-head, 2021-08-09:
No but I did check the pins
I changed the EP-0136 with another one and that I had on hand and that one works perfectly.
Both accepted firmware update but the bad one did not provide enough current to boot the rip only enough to glow the red LED and dimly glow the green

RubenKelevra, 2021-08-15:
I also have a Rasberry PI 4B and received my EP-0136 yesterday. Same issue. Any idea why?

1-prop-head, 2021-08-16:
No

I bought 2 of them and the second one worked pergectly
so it has to be somthing with the unit
Dm

kuba989898, 2021-08-16:
Same problem, Raspberry Pi 4 connected trough spring pins not boot/ stuck at boot loop. Connected additionally USB A port from UPS to USB C port of Raspberry Pi and raspberry started, but is wrong Raspberry pi current power measurment.

Searched at the board and supply line that is going to USB A/C ports going to MOSFET and after MOSFET to the 0.01 ohm resistor and next to the spring pins. Probably MOSFET fault or something. I will investigate it if I discover something share with you.

Hamper, 2022-01-06:
I have same issue, battery voltage is 4.18V.
`i2cdetect -y 1` result:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- 17 -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: 40 -- -- -- -- 45 -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
```

Hamper, 2022-01-09:
I tried to measure voltage and current on 5V pins and get 5.2V without load and max 0.17W (1.7V 0.1A) with some load on this pins, but it must be at least 15W (5V, 3A). I think that some elements in power circuit is broken in my module (RTC and power measurement I2C circuits works fine).

yoyojacky, 2022-01-17:
> I tried to measure voltage and current on 5V pins and get 5.2V without load and max 0.17W (1.7V 0.1A) with some load on this pins, but it must be at least 15W (5V, 3A). I think that some elements in power circuit is broken in my module (RTC and power measurement I2C circuits works fine).

you mean output current max 0.1 A ? seems like the power managment chip is out of work, please contact with sale and get refund or replace part.

closed 2024-09-18 by yoyojacky