geeekpi/upsplus #73: Reboot command locks up unit

ralbright-wa, 2021-08-11:
I was working on mounting a USB drive and needed to reboot the unit.... now the unit will not respond and the function button does nothing. What can be done?

ArjenR49, 2021-08-11:
Not normal at all. When I reboot the Pi on my UPS plus, it reboots without problems. No button needed.
Maybe you shut the Pi down and have some odd settings in your UPS plus?
The ultimate way to get out of such a situation is to take the batteries out and put them back in (without messing up polarity). And then study how you need to set up and operate your UPS. It is not necessarily obvious or easy to understand and do.
You can also do a factory reset (register 0x1B per the documentation). The documentation is also not very easy to grasp.
Or  ... the UPS board could be defective, of course .... It happens, too.

nickfox-taterli, 2021-10-14:
The reboot is executed with the normal reboot command, check first that the reboot command does not get stuck.

closed 2024-09-18 by yoyojacky