geeekpi/upsplus #74: Issues with Current Value and Charging

savejeff, 2021-08-17:
Hi, 

i have found some issues:

first the current readings from both supply and batt in the python script are incorrect. it says the Raspberry consumes 2000mA @5.0V resulting in 10W of power. the batt current also seemed to high. i checked it with a multimeter and both supply and batt current are of by about a factor of 2x. 
```
ina_supply = INA219(0.00725, address=0x40)
ina_batt = INA219(0.005, address=0x45)
```
I think here the wrong resistor values are used. i get roughly the correct values by using double the current values. I think this should be checked.

Second,  I observed frequent interruptions while charging. i plotted the charge current and it drops every 2 minutes. Also right after charging start Power Delivery 2 is used with 9V 3A for 2 minutes then it stops charging and goes to 5V 1.5A. As a result, charging takes longer then discharging. that's not fast charging.

One Discharge and charge cycle
[image]

Start of Charging: 

[image]

ArjenR49, 2021-08-17:
Re the INA219 values:I have noticed the same already in june 2021 and adjusted the parameters roughly based on measured input power in the appropriate situation (referring to your second point).

My INA219 calls:
#        inaBattery = INA219(0.005, busnum=DEVICE_BUS, address=0x45)
        inaBattery = INA219(0.011, busnum=DEVICE_BUS, address=0x45)

and

#        inaRPi = INA219(0.00725, busnum=DEVICE_BUS, address=0x40)
        inaRPi = INA219(0.0145, busnum=DEVICE_BUS, address=0x40)

In the case of your second point my explanation is the battery sampling done by the f/w at regular intervals (default 2 minutes). The blue leds are off while the battery is switched to power the load and the charging power supply is cut off for a number of seconds.
This is somehow documented: if your power supply is 'weak', you are advised to lengthen the sampling interval.
Under maximum (or typical, whatever you prefer) check the battery voltage and the estimated battery capacity. If they drop, your power supply is 'weak' it cannot keep up with the discharging during the battery sampling event. You may be able to remedy that (a little) by changing the ratio of battery sampling time to charging time, i.e. increasing the interval time from 2 minutes.

Re your last point I have not much to add. I never tried to use any of the USB-ports on the shots side until just now ... It charges my phone ... the phone lives ... (USB A). Tried a Xiaomi power bank as well. It charges .. according to the led indicator at least (blinking).

It looks like my UPS f/w is frozen again. At least the minute counts (originally seconds, I converted them in my reporting script) aren't increasing any longer. This is the major problem with this UPS on f/w 9 for me. Not reliable as a UPS. When it freezes, the control scripts cannot shut the Pi down after an AC failure. etc. etc. See other issues for this BUG.
If it hasn't frozen, it's OK.

savejeff, 2021-08-17:
as the parameter of the INA is the shunt resistor value, it should be something round like 0.01 and 0.015. don't really understand how this was not found already. 

about the charging, why must the charging be stoped in order to check the battery state. if I use the UPS as intended where it its allweays plugged into a charger it would charge and discharge the battery at 100% every 2 minutes. that's a good way to age a lithium cell very quickly
I'll try to set the interval to a higher value. what does this interval thing do anyway?

about the charge port, I meant that when I plug a powerbank into the intended charge port on the LONG side it starts to charge the powerbank not the UPS. So there is no way to charge the UPS using a type c powerbank that can be charged and discharged over the same type c port.

I'm considering switching over to the PiSugar. This is too much unexpected behavior for my linking.

hellresistor, 2021-12-12:
> It looks like the resistors used for shunts are 0.01R **R010** 0.01 Ohm 1% 2512 (1W) SMD, so you should use INA219(0.01, ... In my case it 100% fits in measurements with voltage and power meters on battery and Rpi.

Friend, are you saying We should change on python code with that values "0.01"?

Siegurd01, 2021-12-12:
@hellresistor, yep.

ArjenR49, 2022-06-01:
To those worrying about maximum battery capacity % readings from the UPS:

I have changed the USB cable from the charger to the UPS to one by UGREEN advertised for fast charging (AliExpress: Ugreen Usb Kabel PD100W Usb C Naar Type C Fast Charger Kabel Voor Xiaomi Samsung Macbook Ipad 5A Mobiele Telefoon cord Usb-kabel Type C) and the charger to a UGREEN GaN 65 W charger, model CD244, also from AliExpress), which provides up to 3 A at 9 V (on my UPS).
Initial charge current was about 3A when the batteries had been run down, and the charging/battery capacity % rose to 99%, whereas on a 9V/2.22A PD Fast Charger from Ugreen and another USB cable it never got above 97%, more likely 94%.

A more powerful charger and better USB cable indeed seems to make a difference, although I do not think this is essential to the operation of the UPS. Not all PD chargers and USB cables are created equal ....
The new cable and charger were far more expensive than the UPS when I bought it ...

yoyojacky, 2024-09-18:
Hi ArgenR49， That's right. better power supply will offer better perfornance...

closed 2024-09-18 by yoyojacky