geeekpi/upsplus #76: Full battery voltage and capacity incorrect

cat007mine, 2021-08-18:
Hi all.
When I run the included "Full-featured-demo-code.py" script I get the below:

Raspberry Pi power supply voltage: 5.184 V
Current current consumption of Raspberry Pi: 1024.727 mA
Current power consumption of Raspberry Pi: 5318.755 mW
Batteries Voltage: 4.196 V
Battery current (charging), rate: 58.049 mA
Current battery power supplement: 248.780 mW
Current processor voltage: 8147 mV
Current Raspberry Pi report voltage: 4980 mV
Current battery port report voltage: 3936 mV
Current charging interface report voltage (Type C): 5265 mV
Current charging interface report voltage (Micro USB): 48 mV
Currently charging through Type C.
Current battery temperature (estimated): 52 degC
Full battery voltage: 4867 mV
Battery empty voltage: 3700 mV
Battery protection voltage: 3700 mV
Battery remaining capacity: 31 %
Sampling period: 2 Min
Current power state: normal
Shutdown countdown: 240 sec
Automatically turn on when there is external power supply!
Restart countdown: 10 sec
Accumulated running time: 6618 sec
Accumulated charged time: 6625 sec
This running time: 6616 sec
Version number: 9

As you can see, full battery voltage of 4867mV is not correct - this should read 4200mV - and also the battery remaining capacity at 31% does not change. I have tried draining the batteries down to 3200mV (which showed 0% on the above script) and charging them for 24 hours, but although the voltage of the battery goes up to 4.2V, the capacity won't go above 31%.....

I don't need all the details in that script. Is someone able to amend the Full script to just show a few outputs?

All I would like is the below:

Current current consumption of Raspberry Pi: 
Batteries Voltage: 
Battery current (charging), rate: 
Battery empty voltage: 
Battery protection voltage: 
Battery remaining capacity:

Cheers!
Hunt

ArjenR49, 2021-08-19:
Could it be that the Full battery voltage value which is way too high is 
set by the f/w when the user inserts the batteries with the power supply 
already connected? Just a wild guess ...

Perhaps it would be a good idea to reset the UPS to factory defaults and 
start anew.

If you need a script for enabling the factory reset, I have one on my 
Github (https://github.com/ArjenR49/UPS-Plus - there are some useful 
scripts there, although in general it's a bit of a mess, sorry).

There's also my script for reporting on the state of the UPS and 
batteries, but it has a lot of info, too ;-) 
(UPS_report_for_UPSPlus_mqtt.py). Don't worry about the mqtt. It has 
nothing to do with that script.

Enable the factory reset, shut down Pi (from GUI), disconnect power 
supply, remove batteries, reinsert batteries (with correct polarity!): 
Pi starts up. Reconnect power supply. Check register values ...

cat007mine, 2021-08-19:
Thank you! That's great!

I'd like to set my cut off voltage to 3200mV - when I run your UPS_report_for_UPSPlus_mqtt.py I get the below (I've currently got the batteries removed to test run down time - the % went from 99% to 74% almost instantly and dropping really fast).
How do I edit the cut off voltage, which I assume is - "Discharge limit for use by the control script:       3.700 V   (0x11-0x12)" ?

------------------  20-08-2021 11:27:10  ---------------------------------

*** Data from INA219 at 0x40:
Raspberry Pi supply voltage:                         5.070 V
Raspberry Pi current consumption:                    0.588 A
Raspberry Pi power consumption:                      2.850 W

*** Data from INA219 at 0x45:
Battery voltage:                                     4.150 V
Battery current (discharging):                       0.780 A
Battery power consumption:                           3.220 W

*** Remainder of report is based on data collected
*** by the UPS f/w and read from memory at 0x17

UPS board MCU voltage:                               3.280 V   (0x01-0x02)
Voltage at Pi GPIO header pins:                      5.090 V   (0x03-0x04)
USB type C port input voltage:                       0.000 V   (0x07-0x08)
Micro USB port input voltage:                        0.041 V   (0x09-0x0A)

Battery temperature (estimate):                         44°C   (0x0B-0x0C)
Automatic detection of battery type:                   yes     (0x2A)
Batteries fully charged at (UPS/learned value):      4.270 V   (0x0D-0x0E)
Current voltage at battery terminals:                4.150 V   (0x05-0x06)
Discharge limit for use by the control script:       3.700 V   (0x11-0x12)
Batteries fully discharged at (UPS/learned value):   3.700 V   (0x0F-0x10)
Remaining battery capacity (estimate):                  74 %   (0x13-0x14)
Battery sampling ('blue LEDs off') interval:             2 min (0x15-0x16)

Current power state: normal (0x17)

*** EXTERNAL POWER LOST! RUNNING ON BATTERY POWER!

Should the external power be interrupted long enough
to cause the battery voltage to drop below 3.7 V
+ 0.2 V as a safety margin, an appropriate control script
should halt the Pi, after which the UPS
may eventually power it down (& possibly restart it).

UPS power off/on timer registers 0x18 and 0x1A
and register 0x19 should be set to values appropriate
for a power failure event by the control script
immediately before halting the Raspberry Pi.

Current values of the UPS power control registers:
0x18=0 / 0x19=1 / 0x1A=0
Explanation:
0x18 - Power off timer (no restart)   - not set.
0x19 - Automatic restart upon return of external power: yes
0x1A - Power off timer (with restart) - not set.

Accumulated running time:                             1087 min (0x1C-0x1F)
Accumulated charging time:                            1087 min (0x20-0x23)
Current up time:                                      1087 min (0x24-0x27)
F/W version: 9
Serial Number: ************************

Cheers!
Hunt

ArjenR49, 2021-08-20:
It is what Geeekpi calls protection voltage.  The hexadecimal addresses at
the end of the lines refer to the table describing the registers in the
Geeekpi document ( .... EP-0136 ...)
They are the very addresses that the printed value is read from as you can
see in the script code. No magic.

I've just given the register values different names in an attempt to
describe their purpose better. It's a matter of personal preference of
course.

There's a script to set the 'protection voltage' on my GitHub, if I am not
mistaken.
It shows the current value and if you give it an argument, that argument is
set as the new value.
I prefer to run the scripts in Thonny. You can open a field to enter
arguments.
If you make changes to the script and run it, it is stored automatically.

Arjen
(On The Road)

cat007mine, 2021-08-20:
Oops - sorry I didn't check your github properly - have run your protection voltage script and it's set it perfectly - thanks!

Cheers will check out Thonny!

cat007mine, 2021-09-16:
Is there a way to shut the Pi and then then UPS down? If you shut the Pi down - but it still applies power to the Pi so there's no way to start it back up again without pulling the AC power from the UPS and the batteries....

ArjenR49, 2021-09-16:
It should be possible to press the button once and the Pi will start up 
again.

All in all, the content of the following registers will also influence 
what the UPS does. You might want to read up about those and experiment. 
I would suggest you run GitHub Frtz13's control script 
fanShutDownUps.py, and upgrade the UPS to f/w 10 if you haven't done 
that yet.

 From the output of my (Github ArjenR49) report script I copied the 
following info. 0x18 and 0x1A should only be set immediately before 
issuing a shutdown command on the Pi (issued by the control script when 
the battery is below the protection voltage (+ 200 mV)).

Current values of the UPS power control registers:
0x18=0 / 0x19=1 / 0x1A=0
Explanation:
0x18 - Power off timer (no restart)   - not set.
0x19 - Automatic restart upon return of external power: yes
0x1A - Power off timer (with restart) - not set.

When I manually issue a shutdown command in a VNC session from the GUI, 
the Pi shuts down. If I then press the UPS's button, the red LEDs go off 
for a split second after which they come back on and the Pi starts up 
again (because automatic restart upon return of AC is set).

This is with AC charger connected and powered. In case the AC is 
disconnected, the behaviour after shutting down and pressing the button 
is different:

After shutting down, pressing the button will switch off both red LEDs. 
After some delay the blue leds will also turn off. I take it that means 
the UPS uses only minimal power. After pressing the button again, the Pi 
starts up (from battery only, no AC).

ArjenR49, 2021-09-16:
BTW, I tested all the above as I wrote it. Just to make sure I don't tell stories ;-)

onxblog, 2021-11-26:
For manual setup Full battery voltage and Battery empty voltage
You need to change(0x2A)  i2cset -y 1 0x17 0x2a 1 (Enable Battery Parameters self-programmed by user)
Then use script:
import os
import time
import smbus2
import logging
from ina219 import INA219,DeviceRangeError

DEVICE_BUS = 1
DEVICE_ADDR = 0x17
FULLVOLTAGE = 4200
EMPTYVOLTAGE = 2750

bus = smbus2.SMBus(DEVICE_BUS)

bus.write_byte_data(DEVICE_ADDR, 42,1)

print("Set Full Voltage register to: %d" % FULLVOLTAGE)
bus.write_byte_data(DEVICE_ADDR, 13, FULLVOLTAGE & 0xFF)
bus.write_byte_data(DEVICE_ADDR, 14, (FULLVOLTAGE >> 8)& 0xFF)
print("Set Empty Voltage register to: %d" % EMPTYVOLTAGE)
bus.write_word_data(DEVICE_ADDR, 15, EMPTYVOLTAGE)

print("FixedVbat register is: %d" %  bus.read_byte_data(DEVICE_ADDR, 42))

print("Full Voltage register is: %d" % bus.read_word_data(DEVICE_ADDR, 13))
print("Empty Voltage register is: %d" % bus.read_word_data(DEVICE_ADDR, 15))

closed 2024-09-18 by yoyojacky