geeekpi/upsplus #77: Help! How to reset to factory defaults?

ArjenR49, 2021-09-11:
I've done it before, but forgot how and I cannot find instructions.
I have set 0x1B to 1 and then what comes next?
I've restarted the Pi, taken out the batteries and reinserted them several times, also while keeping the button pushed while reinserting. The Pi starts up every time, but it fails to read any i2cbus memory:
`Traceback (most recent call last):
  File "/home/pi/UPSPlus_MQTT/UPS_report_for_UPSPlus_mqtt.py", line 82, in <module>
    aReceiveBuf.append(bus.read_byte_data(DEVICE_ADDR, i))
  File "/home/pi/.local/lib/python3.7/site-packages/smbus2/smbus2.py", line 433, in read_byte_data
    ioctl(self.fd, I2C_SMBUS, msg)
OSError: [Errno 121] Gindse invoer-/uitvoerfout

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/UPSPlus_MQTT/UPS_report_for_UPSPlus_mqtt.py", line 85, in <module>
    raise Exception("[UPS_report] Error reading UPS registers: " + str(e))
Exception: [UPS_report] Error reading UPS registers: [Errno 121] Gindse invoer-/uitvoerfout
pi@RPI-HUB:~/UPSPlus_MQTT $ 
`

I2cdetect -y 1 shows this:
```
pi@RPI-HUB:~/UPSPlus_MQTT $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- 18 -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: 40 -- -- -- -- 45 -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@RPI-HUB:~/UPSPlus_MQTT $ 

```
I seem to remember the number on the first line should be 17 for normal operation.

Where can I find proper instructions for a factory defaults reset?

frtz13, 2021-09-14:
Did you get this sorted out? As far as I remember, the UPS communicates at address 0x18 when in software upgrade mode (you probably got there putting back in the batteries while holding the push button). How to get out of there? one way should be running the firmware upgrade script.

ArjenR49, 2021-09-14:
No, I didn't get out of it. Not yet. I stopped trying, since I was more exhausted by that 'UPS' than its batteries ever were ;-)
But, you are probably right. I haven't yet dismantled the setup, so I now have a way to proceed. The upgrade has always succeeded fine.
THANK YOU.
In the mean time, I had another look at the minefield that UPS's for Raspberry Pi are ... and just an hour ago ordered a UPS Pico HV4.0B Advanced from Greece. I was so fed up ...
If the freezing of the Chinese UPS isn't all sorted out by November, I can probably use the Greek one ;-)
Maybe it is documented better, although I have yet to see any manual for that one ...
I'll try the upgrade later in the evening, probably, and then I can close this issue. I'll let the UPS run with your latest script and nothing else, and not touch it for a week or more ...

ArjenR49, 2021-09-14:
Back to normal ... The UPS was indeed in the upgrade mode. Upgraded smoothly as always ... to version 10 !

After that some restarting done and had to run my own reporting script to check various values.
Surprisingly at first the protection voltage and the fully discharged voltage were both at exactly 0,000 V; never seen such values before. I didn't think this very safe for learning cycles, so I set the protection voltage to 3200 mV, which then resulted in the following values (they sometimes end up differently):
Discharge limit for use by the control script:             3,070 V   (0x11-0x12)
Batteries fully discharged at (UPS/learned value):   3,070 V   (0x0F-0x10)

All in all the UPS seems to be back to normal. Next run down batteries a few times and recharge to have the UPS learn about the batteries.
For now I'll leave the bus speed at 100.000, the default.

Any new features/less problems with f/w 10?
(The GitHub site still advertises version 9)

closed 2021-09-15 by ArjenR49

jeremytedwards, 2022-05-11:
What if this happens and the device is already on v10. Is there a way to get the device out of OTA mode, or reset it to get back to base operation? Running Method 2 "i2cset -y 1 0x17 50 127 b" has a write fail so presume the device is locked up for now.

```
pi@hampi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- 18 -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: 40 -- -- -- -- 45 -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --  

pi@hampi:~ $ i2cset -y 1 0x17 50 127 b
Error: Write failed

```