geeekpi/upsplus #79: Stack Function

alpapak, 2021-09-19:
Can someone help me with the stack function.

Part number to order or any help. I would like to add 2 extra 18650 batteries.

nickfox-taterli, 2021-09-19:
This issue was transferred to the R&D team and flagged.

 
From: alpapak
Date: 2021-09-19 13:33
To: geeekpi/upsplus
CC: Subscribed
Subject: [geeekpi/upsplus] Stack Function (#79)
Can someone help me with the stack function.
Part number to order or any help. I would like to add 2 extra 18650 batteries.

nickfox-taterli, 2021-10-14:
It is possible to connect more lithium batteries via the STACK func, please note that the parameters of the connected lithium batteries should be the same, due to the current global chip tension, the stackable extensions are not available at the moment.