geeekpi/upsplus #80: How to set battery capacity, Full voltage, low voltage?

sddwt, 2021-09-20:
Hi, I follow the code in <https://github.com/geeekpi/upsplus/issues/16>,
After changeing the parameters, and reboot the the UPS, I got error parameter singal (No charging but blue led blinks).
And I get a different value not as I set.
So could anyone providers a valid way to set capacity and voltage? Thanks. 

Code is here:
```
import smbus2

DEVICE_BUS = 1
DEVICE_ADDR = 0x17
FULLVOLTAGE = 4200
EMPTYVOLTAGE = 3300

bus = smbus2.SMBus(DEVICE_BUS)

bus.write_byte_data(DEVICE_ADDR, 42,1)

print("Set Full Voltage register to: %d" % FULLVOLTAGE)
bus.write_byte_data(DEVICE_ADDR, 13, FULLVOLTAGE & 0xFF)
bus.write_byte_data(DEVICE_ADDR, 14, (FULLVOLTAGE >> 8)& 0xFF)
print("Set Empty Voltage register to: %d" % EMPTYVOLTAGE)
bus.write_word_data(DEVICE_ADDR, 15, EMPTYVOLTAGE)

print("FixedVbat register is: %d" %  bus.read_byte_data(DEVICE_ADDR, 42))

print("Full Voltage register is: %d" % bus.read_word_data(DEVICE_ADDR, 13))
print("Empty Voltage register is: %d" % bus.read_word_data(DEVICE_ADDR, 15))
```

ArjenR49, 2021-09-20:
AFAIK the full voltage cannot be set although the code may execute without
error (I don't remember). It is learned by the ups through charging and
discharging the batteries.

Empty voltage is also learned by the ups in the same way. It stays below
the protection voltage by about 100 mV during discharging.

The full voltage stays above the battery voltage by some 100 mV while
charging and then stays at the highest value.

Arjen
(On The Road)

sddwt, 2021-09-20:
> AFAIK the full voltage cannot be set although the code may execute without error (I don't remember). It is learned by the ups through charging and discharging the batteries. Empty voltage is also [...]

Thanks for reply, But I got this error on the page:
> The battery can't charging, Please replace the battery or power adapter.
The battery is new, but I can hardly charge it. And I use QC. And I wait for hours, but still this message.

yoyojacky, 2021-10-14:
@nickfox-taterli  Ok，It‘s done~
https://wiki.52pi.com/index.php/EP-0136#FAQ

closed 2021-10-14 by yoyojacky

sfatula, 2022-12-31:
Dumb me. Left on a trip, thought, you never know, maybe storms whatever so I think I'll leave the power input to the UPS disconnected to protect ot while I am gone. Came back, you guessed it, won't boot! Came here to discover the issue and remedy as I don't have a charger for the batteries other than the UPS! Won't do that again! Since it's part of a case they make, a lot of effort to disconnect.