geeekpi/upsplus #81: OTA script not working , no response from gateway(URL)

evertide, 2021-09-30:
The OTA update script is not working and is giving JSON errors. I suspect that this is due to the fact the URL https://api.52pi.com/update is not working ( 502 bad gateway error).  (https://api.52pi.com is not working either).

error code I am getting
 python3 OTA_firmware_upgrade.py
Traceback (most recent call last):
  File "OTA_firmware_upgrade.py", line 32, in <module>
    r = json.loads(r.text)
  File "/usr/lib/python3.7/json/__init__.py", line 348, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.7/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.7/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

The UPS worked fine prior to the update attempt and setting in thru method 2 (i2cset -y 1 0x17 50 127 b) in upgrade mode.

Is there a i2cset command on 0x18 to switch back to normal mode or is there an alternative URL for the upgrade?

tobylaw, 2021-10-01:
Just to confirm that I too am seeing this same error when trying to upgrade my firmware.

evertide, 2021-10-01:
There is a slight change this morning. The website api.52pi.com is up again but the  /update still gives an error. 

--
Oops! An Error Occurred
The server returned a "405 Method Not Allowed".
Something is broken. Please let us know what you were doing when this error occurred. We will fix it as soon as possible. Sorry for any inconvenience caused.
---

It would be great to have the OTA script changed to just restore it to normal operations ( i2c  address 17)

ArjenR49, 2021-10-01:
IoT reporting in a control script also currently results in an error (copied from HA):
`<html> <head><title>502 Bad Gateway</title></head> <body> <center><h1>502 Bad Gateway</h1></center> <hr><center>nginx/1.18.0</center> </body> </html>
`

ArjenR49, 2021-10-03:
IoT server appears to be up again. IoT reporting from Home Assistant is working on my UPSplus now.

tobylaw, 2021-10-03:
Thanks ArjenR49 for the heads up, yes it is back up and is now version 10. I just download the scripts from github again and no mention of version 10 in the docs. It worries me a little that there is no offline update option for the failed update with the site being down left my UPSplus in a non functioning state. 
We are planning on using these in a number of our products and not being able to reliably installed updates could be a real headache, especially if there is no way of rolling back.

ArjenR49, 2021-10-03:
Version 10 blocks i2c bus access from the pi at regular intervals.
Depending on the sampling interval setting.

That doesn't prevent operation of frtz14's control script in my experience.
I have not used Geeekpi's control script for s long time now.

Arjen
(On The Road)

evertide, 2021-10-04:
I can confirm the upgrade worked hence the upgrade servers is back up and running. I still think it would be good to have a bailout option in the software changing it back to operation ( (17) mode. But that would be more of a feature request.

closed 2021-10-04 by evertide