geeekpi/upsplus #91: Shutdown countdown in minutes?

kralverde, 2021-11-29:
It appears the shutdown countdown is in minutes rather than seconds (as per https://wiki.52pi.com/index.php/EP-0136) in firmware v10

Is there a change log for changes like this?

frtz13, 2021-11-30:
with a value of 60 at register 0x18, my UPS shuts down in one minute, as specified (with fw v10).

kralverde, 2021-11-30:
I will do more testing. Is there any possible reason for this behavior change? Is there a more in-depth register map than the one on 52pi.com?

ArjenR49, 2022-06-08:
In f/w v.10 countdown timer at 0x1A will count down and cut the power. Then the UPS waits 10 minutes before restoring power. I haven't seen this documented so one could easily think nothing is going to happen, but it will ... after 10 minutes. I don't know if the 10 minutes can be changed.