geeekpi/upsplus #92: Keyboard connected to USB ports

hellresistor, 2021-12-15:
hello friends.

Today I have try connect my keyboard to USB port.

Keyboard its powered. but not detecting on OS.

Some trick here? o0

My boot config file:
```
root@FIFUrpi:~# cat /boot/firmware/config.txt
# Switch the CPU from ARMv7 into ARMv8 (aarch64) mode
arm_64bit=1

core_freq=
enable_uart=1
upstream_kernel=1

dtparam=i2c_arm=on
dtparam=spi=on
dtoverlay=i2c-rtc,ds1307,addr=0x68
dtparam=i2c_vc=on
dtoverlay=dwc2,dr_mode=host

kernel=vmlinuz-5.10.0-9-arm64
# For details on the initramfs directive, see
# https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=10532
initramfs initrd.img-5.10.0-9-arm64
```

```
root@FIFUrpi:~# cat /etc/modules-load.d/rpi4.conf
i2c-dev
rtc-ds1307
i2c-bcm2835
i2c-smbus
dwc2
```

Pi-Mania, 2021-12-23:
The UPS board supplies Power (& RTC in latest models) no passthru for a keyboard/mouse to work. (Unless your powering a BT keyboard/ mouse)

hellresistor, 2021-12-23:
ok, not have a BT keyboard mouse at moment to confirm.

When Possible I post here the results :) 
Thank you

closed 2021-12-23 by hellresistor