geeekpi/upsplus #93: RaspBerry Pi4 with EP-0136 won't boot

RafaelMuniz94, 2021-12-16:
So I installed the EP-0136 in my pi 4 and it won't boot if the usb c (on the pi board) isn't connected. Even if I connect in the usb c in ups board it won't boot. The ACT light steel blink and only stops when I connect in the pi usb c port.

OAT update done
Batteries fully charged
All GPIO connections are made

Thanks in advance

hellresistor, 2021-12-20:
and try click on Yellow button on UPS?

Pi-Mania, 2021-12-23:
What does >sudo i2cdetect -y 1   display?

I have spent a week playing around with these boards. Definitely UPDATE, (had lots of issues till i did) and >sudo i2cset -y 1 0x17  25 1 (to set auto start on AC inserted function)

Lonewalker0007, 2022-01-17:
I have the same issue. But the when i power the pi via the ups the green light on the pi flashes 2 times. Indicating it cant read the sd card. But at the moment i boot the pi via its own usb C it starts just up.

hellresistor, 2022-01-19:
try full charge batterys first

thomtux, 2022-01-19:
I have the same problem, the raspberry will not start. I tried 2 Different sets of battery brands and the problem doesn't disappear. I put the batteries in the ups and measured the voltage at pin 1 and pin 6. The value is 5.02 V. For me it says that the batteries are fine. They do not collapse under 5 V, what could be to low for functionallty.
In the end the rpi wont start with this ups, doesn't matter if the batteries are fully loaded and and the ups is swited to a power supply.