geeekpi/upsplus #94: batt_current = ina_batt.current()

DEAWid, 2021-12-29:
Can someone help me?

Thanks :)

Error:

Traceback (most recent call last):
  File "/home/pi/Desktop/test.py", line 37, in <module>
    batt_current = ina_batt.current()
  File "/usr/local/lib/python3.7/dist-packages/ina219.py", line 196, in current
    self._handle_current_overflow()
  File "/usr/local/lib/python3.7/dist-packages/ina219.py", line 246, in _handle_current_overflow
    self._increase_gain()
  File "/usr/local/lib/python3.7/dist-packages/ina219.py", line 271, in _increase_gain
    raise DeviceRangeError(self.__GAIN_VOLTS[gain], True)
ina219.DeviceRangeError: Current out of range (overflow), for gain 0.32V, device limit reached

[image]
[image]
[image]

test.py file:
#!/usr/bin/env python3

# Description：
'''This is the demo code for all the functions of UPS Plus.
Advanced users can select the functions they need through the function options provided in the code below to customize and develop them to meet their needs.
'''

import time
import smbus2
import logging
from ina219 import INA219,DeviceRangeError
from pprint import pprint

DEVICE_BUS = 1
DEVICE_ADDR = 0x17
PROTECT_VOLT = 3700
SAMPLE_TIME = 2

ina_supply = INA219(0.00725, busnum=DEVICE_BUS, address=0x40)
ina_supply.configure()
supply_voltage = ina_supply.voltage()
supply_current = ina_supply.current()
supply_power = ina_supply.power()
print("Raspberry Pi power supply voltage: %.3f V" % supply_voltage)
print("Current current consumption of Raspberry Pi: %.3f mA" % supply_current)
print("Current power consumption of Raspberry Pi: %.3f mW" % supply_power)

ina_batt = INA219(0.005, busnum=DEVICE_BUS, address=0x68)
pprint(ina_batt.configure())
ina_batt.configure()

batt_voltage = ina_batt.voltage()
batt_current = ina_batt.current()
batt_power = ina_batt.power()
print("Batteries Voltage: %.3f V" % batt_voltage)
try:
    if batt_current > 0:
        print("Battery current (charging), rate: %.3f mA" % batt_current)
        print("Current battery power supplement: %.3f mW" % batt_power)
    else:
        print("Battery current (discharge), rate: %.3f mA" % batt_current)
        print("Current battery power consumption: %.3f mW" % batt_power)
except DeviceRangeError:
    print('Battery power is too high.')

bus = smbus2.SMBus(DEVICE_BUS)

aReceiveBuf = []
aReceiveBuf.append(0x00)   # Placeholder

for i in range(1,255):
    aReceiveBuf.append(bus.read_byte_data(DEVICE_ADDR, i))

print("Current processor voltage: %d mV"% (aReceiveBuf[2] << 8 | aReceiveBuf[1]))
print("Current Raspberry Pi report voltage: %d mV"% (aReceiveBuf[4] << 8 | aReceiveBuf[3]))
print("Current battery port report voltage: %d mV"% (aReceiveBuf[6] << 8 | aReceiveBuf[5])) # This value is inaccurate during charging
print("Current charging interface report voltage (Type C): %d mV"% (aReceiveBuf[8] << 8 | aReceiveBuf[7]))
print("Current charging interface report voltage (Micro USB): %d mV"% (aReceiveBuf[10] << 8 | aReceiveBuf[9]))

if (aReceiveBuf[8] << 8 | aReceiveBuf[7]) > 4000:
    print('Currently charging through Type C.')
elif (aReceiveBuf[10] << 8 | aReceiveBuf[9]) > 4000:
    print('Currently charging via Micro USB.')
else:
    print('Currently not charging.')   # Consider shutting down to save data or send notifications 

print("Current battery temperature (estimated): %d degC"% (aReceiveBuf[12] << 8 | aReceiveBuf[11])) # Learned from the battery internal resistance change, the longer the use, the more stable the data.
print("Full battery voltage: %d mV"% (aReceiveBuf[14] << 8 | aReceiveBuf[13]))
print("Battery empty voltage: %d mV"% (aReceiveBuf[16] << 8 | aReceiveBuf[15]))
print("Battery protection voltage: %d mV"% (aReceiveBuf[18] << 8 | aReceiveBuf[17]))
print("Battery remaining capacity: %d %%"% (aReceiveBuf[20] << 8 | aReceiveBuf[19]))  # At least one complete charge and discharge cycle is passed before this value is meaningful.
print("Sampling period: %d Min"% (aReceiveBuf[22] << 8 | aReceiveBuf[21]))
if aReceiveBuf[23] == 1:
    print("Current power state: normal")
else:
    print("Current power status: off")

if aReceiveBuf[24] == 0:
    print('No shutdown countdown!')
else:
    print("Shutdown countdown: %d sec"% (aReceiveBuf[24]))

if aReceiveBuf[25] == 1:
    print("Automatically turn on when there is external power supply!")
else:
    print("Does not automatically turn on when there is an external power supply!")
if aReceiveBuf[26] == 0:
    print('No restart countdown!')
else:
    print("Restart countdown: %d sec"% (aReceiveBuf[26]))

print("Accumulated running time: %d sec"% (aReceiveBuf[31] << 24 | aReceiveBuf[30] << 16 | aReceiveBuf[29] << 8 | aReceiveBuf[28]))
print("Accumulated charged time: %d sec"% (aReceiveBuf[35] << 24 | aReceiveBuf[34] << 16 | aReceiveBuf[33] << 8 | aReceiveBuf[32]))
print("This running time: %d sec"% (aReceiveBuf[39] << 24 | aReceiveBuf[38] << 16 | aReceiveBuf[37] << 8 | aReceiveBuf[36]))
print("Version number: %d "% (aReceiveBuf[41] << 8 | aReceiveBuf[40]))

#The following code demonstrates resetting the protection voltage
# bus.write_byte_data(DEVICE_ADDR, 17,PROTECT_VOLT & 0xFF)
# bus.write_byte_data(DEVICE_ADDR, 18,(PROTECT_VOLT >> 8)& 0xFF)
# print("Successfully set the protection voltage as: %d mV"% PROTECT_VOLT)

#The following code demonstrates resetting the sampling period
# bus.write_byte_data(DEVICE_ADDR, 21,SAMPLE_TIME & 0xFF)
# bus.write_byte_data(DEVICE_ADDR, 22,(SAMPLE_TIME >> 8)& 0xFF)
# print("Successfully set the sampling period as: %d Min"% SAMPLE_TIME)

# Set to shut down after 240 seconds (can be reset repeatedly)
# bus.write_byte_data(DEVICE_ADDR, 24,240)
bus.write_byte_data(DEVICE_ADDR, 24,240)

# Cancel automatic shutdown
# bus.write_byte_data(DEVICE_ADDR, 24,0)

# Automatically turn on when there is an external power supply (If the automatic shutdown is set, when there is an external power supply, it will shut down and restart the board.)
# 1) If you want to completely shut down, please don't turn on the automatic startup when there is an external power supply.
# 2) If you want to shut down the UPS yourself because of low battery power, you can shut down the UPS first, and then automatically recover when the external power supply comes.
# 3) If you simply want to force restart the power, please use another method.
# 4) Set to 0 to cancel automatic startup.
# 5) If this automatic startup is not set, and the battery is exhausted and shut down, the system will resume work when the power is restored as much as possible, but it is not necessarily when the external power supply is plugged in.
# bus.write_byte_data(DEVICE_ADDR, 25,1)
bus.write_byte_data(DEVICE_ADDR, 25,1)

# Force restart (simulate power plug, write the corresponding number of seconds, shut down 5 seconds before the end of the countdown, and then turn on at 0 seconds.)
# bus.write_byte_data(DEVICE_ADDR, 26,30)
bus.write_byte_data(DEVICE_ADDR, 26, 10)

# Restore factory settings (clear memory, clear learning parameters, can not clear the cumulative running time, used for after-sales purposes.)
# bus.write_byte_data(DEVICE_ADDR, 27,1)

# Enter the OTA state (the user demo program should not have this thing, after setting, unplug the external power supply, unplug the battery, reinstall the battery, install the external power supply (optional), you can enter the OTA mode and upgrade the firmware.)
# bus.write_byte_data(DEVICE_ADDR, 50,127)

# Serial Number 
UID0 = "%08X" % (aReceiveBuf[243] << 24 | aReceiveBuf[242] << 16 | aReceiveBuf[241] << 8 | aReceiveBuf[240])
UID1 = "%08X" % (aReceiveBuf[247] << 24 | aReceiveBuf[246] << 16 | aReceiveBuf[245] << 8 | aReceiveBuf[244]) 
UID2 = "%08X" % (aReceiveBuf[251] << 24 | aReceiveBuf[250] << 16 | aReceiveBuf[249] << 8 | aReceiveBuf[248])
print("Serial Number is:" + UID0 + "-" + UID1 + "-" + UID2 )

frtz13, 2021-12-29:
i have the impression that you do NOT have a EP-0136 UPS. the i2cdetect command for this device should show i2c responses for addresses 0x17, 0x40, 0x47 and 0x68, 0x68 being the real time clock.
your device only shows responses for 0x40 and 0x68, and there is a good chance that 0x68 is a real time clock and NOT an INA219 sensor which your code tries to address. there is a good chance that your UPS is some other model (EP-118 ??? check with the photos on the 52pi wiki).
make sure you got the UPS which ordered, and just keep the first part of the .py script which seems to be working.

DEAWid, 2021-12-29:
I'm trying this UPS:
https://www.ebay.com/itm/184306934001

And exactly .. I get some info about voltage from first part in script, but script around battery shows me that error :/

frtz13, 2021-12-30:
so you have a EP-0118 device. you should use the info + demo script from the UPSV3 github repository,