geeekpi/upsplus #96: New device, set up error 121

alpapak, 2022-01-20:
Hello,
I have one upsplus and after following the setup manual and running the install script i got error 121

[image]

Any idea?

If i plug the usb-c on upsplus, one blue light 25% is blinking. if i press the button, the pi is not power up. i can power pi only if i plug the usb-c in pi.

check0r0, 2022-01-22:
Buster or Bullseye?

Put this down in the console: sudo i2cdetect -y 1
and show the result.

alpapak, 2022-01-23:
Hi,

i have Buster. It seems that i have to add in boot/config.txt dtoverlay=i2c-rtc,ds1307. Now, it is boot up and then halt

ekaftan, 2022-05-10:
Did you fix the problem?

I have the same problem right now.

pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- 18 -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: 40 -- -- -- -- 45 -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@raspberrypi:~ $ i2cget -y 1 0x17 25
Error: Read failed

Its a 3B+ running Buster.

mmaga, 2022-10-22:
> Did you fix the problem?
>
> I have the same problem right now.
>
> pi@raspberrypi:~ $ i2cdetect -y 1 0 1 2 3 4 5 6 7 8 9 a b c d e f 00: -- -- -- -- -- -- -- -- -- -- -- -- -- 10: -- -- -- -- -- -- -- -- [...]

I have a RPi 4B model.

I've tried to write 0x00 in the address 50 in the same way OTA_firmware_upgrade.py does at the end of the script without success (bus.write_byte_data(0X18, 50, 0)):

i2cset -y 1 0x18 50 0 b

The only way I could get out of the firmware upgrade mode (0x18) to the operational mode (0x17) was to proceed a complete firmware upgrade running:

python3 OTA_firmware_upgrade.py

I removed batteries and external power source, reinstalled batteries, and the UPSPlus was working in the 0x17 operation mode, back again.

alpapak, 2022-10-22:
i gave it up and order another hat from another company.

mmaga, 2022-10-22:
@alpapak, which one, if you don't mind to inform for a next purchase? because it is very disappointing about having no clue on miss-functions of the upsplus.

alpapak, 2022-10-22:
@mmaga 
please check me repo.
you will find your answer.
i am using it in my project for 6months and i am very happy.