geeekpi/upsplus #98: All provided installation scripts fail with ina219.DeviceRangeError: Current out of range (overflow), for gain 0.32V, device limit reached

wojciechczyz, 2022-02-25:
Traceback (most recent call last):
  File "/home/pi/bin/upsPlus.py", line 41, in <module>
    batt_power = ina_batt.power()
  File "/home/pi/.local/lib/python3.9/site-packages/ina219.py", line 204, in power
    self._handle_current_overflow()
  File "/home/pi/.local/lib/python3.9/site-packages/ina219.py", line 246, in _handle_current_overflow
    self._increase_gain()
  File "/home/pi/.local/lib/python3.9/site-packages/ina219.py", line 271, in _increase_gain
    raise DeviceRangeError(self.__GAIN_VOLTS[gain], True)
ina219.DeviceRangeError: Current out of range (overflow), for gain 0.32V, device limit reached
UPS Plus Installation is Incomplete! ... failed!
Please visit wiki for more information:.
-

I have upgraded firmware to latest version via OTA, yet all scripts fail for example:

pi@raspberrypi:~ $ curl -Lso- https://raw.githubusercontent.com/geeekpi/upsplus/main/install.sh | bash
Welcome to 52Pi Technology UPS Plus auto-install Program!.
More information please visit here:.
-----------------------------------------------------.
https://wiki.52pi.com/index.php/UPS_Plus_SKU:_EP-0136.
-----------------------------------------------------.
Start the configuration environment check....
Network status is ok....
Start the software check....
git has been installed..
pi-ina219 library has been installed.
Installing smbus2 library....
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: smbus in ./.local/lib/python3.9/site-packages (1.1.post2)
Requirement already satisfied: smbus2 in ./.local/lib/python3.9/site-packages (0.4.1)
smbus2 Installation successful..
create /home/pi/bin directory....
Create python3 script in location: /home/pi/bin/upsPlus.py Successful.
Create UPS Plus IoT customer service python script successful.
Add into general crontab list..
crontab has been created successful!.
------------------------------------------------------------
------Current information of the detected Raspberry Pi------
------------------------------------------------------------
Raspberry Pi Supply Voltage: 4.964 V
Raspberry Pi Current Current Consumption: 1164.172 mA
Raspberry Pi Current Power Consumption: 5776.283 mW
------------------------------------------------------------
Traceback (most recent call last):
  File "/home/pi/bin/upsPlus.py", line 41, in <module>
    batt_power = ina_batt.power()
  File "/home/pi/.local/lib/python3.9/site-packages/ina219.py", line 204, in power
    self._handle_current_overflow()
  File "/home/pi/.local/lib/python3.9/site-packages/ina219.py", line 246, in _handle_current_overflow
    self._increase_gain()
  File "/home/pi/.local/lib/python3.9/site-packages/ina219.py", line 271, in _increase_gain
    raise DeviceRangeError(self.__GAIN_VOLTS[gain], True)
ina219.DeviceRangeError: Current out of range (overflow), for gain 0.32V, device limit reached
UPS Plus Installation is Incomplete! ... failed!
Please visit wiki for more information:.
-----------------------------------------------------.
https://wiki.52pi.com/index.php/UPS_Plus_SKU:_EP-0136.
-----------------------------------------------------.
pi@raspberrypi:~ $

I have created separate issue here as well:

https://github.com/chrisb2/pi_ina219/issues/32

hellresistor, 2022-04-09:
try with sudo

wojciechczyz, 2023-01-22:
sudo has not helped, there are no problems accessing tools nor io, simply values read are not correct

hellresistor, 2023-01-23:
what is system you are using? aarch64? debian ?
`uname -a`

wojciechczyz, 2023-01-23:
It is Raspbian light 32 bit.

I2C is enabled, otherwise it would never read this:
Raspberry Pi Supply Voltage: 4.964 V
Raspberry Pi Current Current Consumption: 1164.172 mA
Raspberry Pi Current Power Consumption: 5776.283 mW

Somehow I just think the board may be defective, especially that after several hours even with power supply connected batteries finally go flat and raspberry shut downs. Batteries are doublepow as recomended on board docs