geeekpi/upsplus #106: UPS on PD charger enters fault condition -> batteries discharged to 2,62 V!

ArjenR49, 2022-06-14:
My UPS plus (f/w 10) and Pi4B have been on a UGreen PD charger delivering 9V and over 3 A charging current when the batteries are empty while also running the Pi.
My Pi/UPS plus combination is controlled by Frtz13's fanShutDownUps.py script, which has been performing just fine. The battery capacity reading had been at 99% for a few days and I was deliberately keeping the Pi mostly idle to see if the percentage would ever go to 100% and the last blue LED might stop blinking.

I have had one earlier experience of last nights events, but this time I have more detailed observations.

**This morning I found the Pi totally unresponsive**, even though the charger had been connected all the time since over 24 hrs. It didn't answer to ping nor were any connections possible. At that moment I didn't have a monitor connected.
**Yet all four blue LEDs were on and NONE blinking.**

After disconnecting power to the charger and reconnecting the two red LEDs come on and the Pi apparently tries to start as it is blinking its green LED a few times. 
On a HDMI monitor there was nothing to see, no signal.
After flashing the green LED a few times, the red (power) LEDs went off again for a second or so, after which they came on again and the (failing startup) sequence just repeated itself.
Disconnecting charger power and reconnecting didn't help. It just started over with what looked like a failing startup over and over.
By this time the blue LEDs were off.

Eventually I decided it's time to take the bottom plate off and take out the batteries and measure the battery voltage, because I suspected what happens is that the f/w disconnects the batteries as soon as it comes to life because the voltage is far too low.

Indeed my trusty Amprobe AM-500-EUR digital multimeter showed **2,62 V** for the batteries (no load, outside the UPS).
There was now nothing else to do but put them in a separate Nitecore charger. It also showed 2,62 V and started charging.

Since **the purpose of having a UPS is keeping the Pi running no matter what**, this is real bad.

I have the UPS/charger on a remotely controlled AC switch.

I have now set the switch to do a daily power off lasting 15 minutes. That should keep the batteries from reaching 100%.

Before this fault condition happened, after a factory defaults reset, over the course of several days I had run down and recharged the batteries at least two times by switching the charger off and back on, as soon as the Pi was shut down by the control script fanShutDownUps.py and the UPS cut power to the Pi  by virtue of the countdown timer.
The voltage settings on UPS which it learned during this 'training' were totally normal (max. about 4,3 V and batteries always stayed below that at some 4,2xx V (batt. capacity at 99%).

The Pi's syslog may reveal more of what happened, but first I have to get the batteries charged again.

**I hope Geeekpi's product support will address this issue.**

Pegi68, 2022-10-21:
Same issue too
Batterie was unloaded to 2.6V and if the power is on from an original RPi Power Supply 5V/3A the RPi try to start and start and start... :-(
I removed the sd-card and have to wait for the battery to charge a little bit

jens-roth, 2024-05-13:
Hello, same issue here too. I tried different USB-C-Chargers (UGREEN 65W Charger and the new IKEA 30W Charger) with USB-C and Micro-USB-Input. After a while UPS Plus stops "charging" and switches to discharging mode. Via I2C I read full "normal" Voltage an USB-C or Micro-USB-Port, but the UPS-Plus is silently discharging to the bitter end (under 2,6V).
I tried to change the "Sampling Period" from 60min back to 2min via SSH/Scipt. The UPS Plus then switched to back to charging (blinking blue LED's). This is a very serious issue.

yoyojacky, 2024-07-01:
@jens-roth 
Please refer to wiki's instructions: https://wiki.52pi.com/index.php?titile=EP-0136

yoyojacky, 2024-07-01:
> My UPS plus (f/w 10) and Pi4B have been on a UGreen PD charger delivering 9V and over 3 A charging current when the batteries are empty while also running the Pi. My Pi/UPS plus combination is controlled by Frtz13's fanShutDownUps.py script, which has been performing just fine. The battery capacity reading had been at 99% for a few days and I was deliberately keeping the Pi mostly idle to see if the percentage would ever go to 100% and the last blue LED might stop blinking.
> 
> I have had one earlier experience of last nights events, but this time I have more detailed observations.
> 
> **This morning I found the Pi totally unresponsive**, even though the charger had been connected all the time since over 24 hrs. It didn't answer to ping nor were any connections possible. At that moment I didn't have a monitor connected. **Yet all four blue LEDs were on and NONE blinking.**
> 
> After disconnecting power to the charger and reconnecting the two red LEDs come on and the Pi apparently tries to start as it is blinking its green LED a few times. On a HDMI monitor there was nothing to see, no signal. After flashing the green LED a few times, the red (power) LEDs went off again for a second or so, after which they came on again and the (failing startup) sequence just repeated itself. Disconnecting charger power and reconnecting didn't help. It just started over with what looked like a failing startup over and over. By this time the blue LEDs were off.
> 
> Eventually I decided it's time to take the bottom plate off and take out the batteries and measure the battery voltage, because I suspected what happens is that the f/w disconnects the batteries as soon as it comes to life because the voltage is far too low.
> 
> Indeed my trusty Amprobe AM-500-EUR digital multimeter showed **2,62 V** for the batteries (no load, outside the UPS). Time will tell if the batteries took a hit from being so deeply discharged. There was now nothing else to do but put them in a separate Nitecore charger. It also showed 2,62 V and started charging.
> 
> Since **the purpose of having a UPS is keeping the Pi running no matter what**, this is real bad.
> 
> I have the UPS/charger on a remotely controlled AC switch.
> 
> I have now set the switch to do a daily power off lasting 15 minutes. That should keep the batteries from reaching 100%.
> 
> Before this fault condition happened, after a factory defaults reset, over the course of several days I had run down and recharged the batteries at least two times by switching the charger off and back on, as soon as the Pi was shut down by the control script fanShutDownUps.py and the UPS cut power to the Pi by virtue of the countdown timer. The voltage settings on UPS which it learned during this 'training' were totally normal (max. about 4,3 V and batteries always stayed below that at some 4,2xx V (batt. capacity at 99%).
> 
> The Pi's syslog may reveal more of what happened, but first I have to get the batteries charged again.
> 
> Additional info: Upon reinserting the batteries the Pi started up with those partially recharged batteries (3,8 V), but it suddenly stopped running after about a minute. Not being prepared for this, I wasn't able to check much if anything. I noticed the protection voltage level was at 3000 mV, which is not what it was before the fault condition. It had left it at either 3200 mV or 3400 mV. I've had it at 3000 mV only during the first battery learning sequence.
> 
> The UPS seems to be really mixed up and not showing normal behaviour. It should have allowed the Pi to run as the batteries were clearly above any protection voltage level.
> 
> More additional info: Charged the batteries in an external charger up to over 4 V and then placed them in the UPS. THis time the Pi was able to start. I did a reset to factory defaults and left the batteries to charge. When the batteries were more or less fully charged, with the protection voltage set at a conservative 3400 mV, I disconnected the charger so the UPS would 'learn' about the batteries. By now the UPS had adjusted the learned value to 3550 mV (some 50 mV below 3400 mV + 200 mV margin as set by Frtz13's UPS control program). How exactly the UPS plus determines various values it 'learns' about, is undocumented and subject to speculation, but the upper and lower voltage limits seem to be pushed to about 50-75 mV above and below the actually measured maximum and minimum voltages. Only testing can tell one how the UPS on f/w v.10 reacts to different things the user does to the UPS. Charge, discharge to a lower limit. ... changing the voltage in (0x11-0x12) will also sometimes influence what is in (0x0F-0x10).
> 
> **I hope Geeekpi's product support will address this issue.**

This is obviously a battery problem, please try replacing the battery.

closed 2024-07-01 by yoyojacky