geeekpi/upsplus #109: Error 121 with no 17 registered

MarkStayt, 2022-08-15:
I am having issues installing my ups.  I run the sudo ./install.sh script and I get the folowing error

[image]

When I run the sudo i2cdetect -y 1, I see 40, 45, 68 but no 17..

[image]

Any idea what I can do? 

Purchased on Amazon from GeeekPi Store as part of  ZP-0129 on Aug 9 and received it on the 13th.

Pegi68, 2022-10-20:
I also had purcased two GeeekPi Raspberry Pi UPS from Amazon and had problems to install the second one.
The installation show me "The installation is incompatible" and failed.
If I checked the I2C Address with the i2c-tools and saw the first UPS was address 0x17 and the second had 0x18 !?
I corrected the upsplus.py script and can run it without problems.

Perhaps I could help al little bit.

HS-Tech-kr, 2023-01-07:
I got same problem. There was [Errno121] and no 0x17 after I changed with new batteries.

yoyojacky, 2024-07-01:
It seems like you are entering `OTA mode`, did you push the button while insert the batteries ?

yoyojacky, 2024-07-01:
> I got same problem. There was [Errno121] and no 0x17 after I changed with new batteries. (with the brunt smell from the PCB)

Could you please take a photo on PCB near by the burnt position and send it back to me ?

closed 2024-07-01 by yoyojacky