geeekpi/upsplus #110: Raspberry Pi Zero 2 (Un)supported

eqikkwkp25-cyber, 2022-09-25:
Hello, 

as the 'new' Raspberry Pi Zero 2W is not listed the question is if its supported or not. In the latter case what is the reason that it does not work with UPSPlus?

eqikkwkp25-cyber, 2022-10-29:
Any updates or hints?

galtobellojr, 2022-10-29:
The Pi Zero 2W uses the same GPIO as the Pi 4B, but my concern would be mounting. Two of the 4 standoffs that attach this to the 4B will not fit the Zero 2W (it's narrower) and since it uses POGO pins to connect to the GPIO from below I'd be concerned about how secure the mount is physically, as well as if the pins would align properly & securely. 

So electrically I'd expect no problems, but mounting will be a pain and I would research that when connecting this or any add-on board/HAT made for the 4B to a Pi Zero.

yoyojacky, 2024-04-08:
It can not mouting on Raspberry Pi zero series , sorry for that!

closed 2024-04-08 by yoyojacky