geeekpi/upsplus #112: Battery compatibility

sturla78, 2022-11-07:
Hello,
i have read on documentation (https://wiki.52pi.com/index.php/EP-0136) that this board support Support 4.2V 4.35V 4.4V 4.5V 18650 lithium battery, i have purchase two but 3.7V. Can be used with this board ?

galtobellojr, 2022-11-07:
My experience with the device is small and eight months past, but I believe my batteries were also 3.7V and the device worked, but only after tweaking some of its settings. In particular, I found the low-voltage setting too low, especially since the value used was (iirc) 0.3V above that specified in the code. This resulted in a low-voltage shutdown very close to the batteries' nominal "charged" level, so minimal run time. Ran (on battery) a CPU test for 4 hours with the fan going after that was corrected, much longer with normal loads. 

It was easy to tweak, and the upsplus device worked mostly as expected. The major exception I'd warn about is that when left off of mains power for a couple of weeks, the batteries are drained to 0V. This wouldn't be a problem in "typical" UPS always-powered usage, but I'd hoped to have a portable Pi that could be used continuously on mains and battery for a time, then stored for later. Battery removal seems the only safe way to store the unit for any time. I replaced the battery cover screws with knurled ones to facilitate.

sturla78, 2022-11-07:
Hello @galtobellojr, only one question: you have follow #16 in order to set the min and max voltage limit ?

Thanks

galtobellojr, 2022-11-07:
@sturla78, I just looked at the discussion you linked, don't recall seeing it before (but given my memory...). I don't recall my FW version, but do seem to recall issues with rapid access to the registers. In my experience, the registers accepted values I wrote to them. And (without access to my code just now) I do believe one change I made was to the protection voltage, which I determined needed to be lower than the default. 

I recall dealing with the issues at the time & thinking "this can't be exceptional" given the wide variety (as I discovered) of 18650 batteries. At the time, this was my first use of 18650s, and so my first practical glimpse of the LiIon hellscape. Most of my problems centered around the lack of documentation for this device, unawareness of the GitHub threads, and my newness to the Raspberry/Python world; the changes iirc were trivial. (...once I realized "18650" batteries can have such different specs; it wasn't me causing the batteries to drain to 0V; there was a cron task overwriting my register settings, etc...)

I have seen a thread here about older (v4?) FW versions stuck in reboot cycle when returned to mains, and observed (but don't think anyone commented) that the "mysterious" change in v10(?) that went from 30sec to several mins delay before power-restore-reboot might have been a response to that (prevent cycling by giving drained batts a few minutes' charge before adding load).

It all seemed pretty straightforward to this Raspberry novice with 40yrs in the industry, once I self-documented all the mismatched moving parts. Don't recall seeing any bugs (though that damned cron task made it look like my register writes weren't "taking").

sturla78, 2022-11-07:
Thanks again galtobellojr, i will try to write a new value for low voltage and protection level as described on the issue 16, and then see what happen.

Me too strating to have many years and if i can i try the simple solution :) 😀

closed 2024-04-08 by yoyojacky